rust-embedded/cortex-m-rt #5: Support a heap memory region

japaric, 2017-04-28:
The obvious place to place this region is right after the `.data + .bss` region. Should we go ahead with that or should we support placing it at a custom address?

whitequark, 2017-04-29:
That sounds good as a first approximation, and doesn't have any critical issues I can think of.

japaric, 2017-04-30:
FWIW, you can already create a heap with the current version of this crate. The `_edata` symbol already points to the end of .data so you can use it as the start of the heap.

Here are [some instructions](https://gist.github.com/japaric/7fdd6b7c1973f605f6903aa7513b7f21) on how to use the alloc-cortex-m crate to allocate a `Vec`. The instructions assume a cortex-m-quickstart setup.

I think the next steps for this issue would be

- create a proper `_heap_start` symbol that is just an alias of `_edata`
- write documentation about the `_heap_start` symbol here
- add an example to cortex-m-quickstart that shows how to initialize and use an allocator

whitequark, 2017-04-30:
Yeah, I've seen alloc-cortex-m. I actually have no need in these instructions (at least right now) because all of my code (especially my TCP/IP stack) are designed to be heap-less :)

japaric, 2017-05-26:
I have seen some linker scripts declare a `.heap` section in the linker script. Has Anyone any need for that? I can't think of any use for such section myself.

whitequark, 2017-05-26:
It is quite useless.

PR #11 (2017-05-27): add a _sheap symbol where a heap can be located
closes #5

I'll merge this after I write some docs for cortex-m-quickstart
commit d0d13678 (2017-05-27): add a _sheap symbol where a heap can be located

closes #5
closed 2017-05-27 by japaric