rust-embedded/cortex-m-rt #12: .bss allocated outside RAM

kjetilkjeka, 2017-05-28:
When using cortex-m-rt .bss is allocated outside RAM which again causes hard fault when reset_handler calls zero_bss.

The microcontroller in use is NXP S32K144

The memory.x file looks like the following:
    MEMORY
    {
      /* NOTE K = KiBi = 1024 bytes */
      /* TODO Adjust these memory regions to match your device memory layout */
      FLASH : ORIGIN = 0x00000000, LENGTH = 512K
      RAM : ORIGIN = 0x1FFF1000, LENGTH = 60K
    }
    
    /* This is where the call stack will be allocated. */
    /* The stack is of the full descending type. */
    /* NOTE Do NOT modify `_stack_start` unless you know what you are doing */
    _stack_start = ORIGIN(RAM) + LENGTH(RAM);

The ram on s32k series ends at 0x1FFF FFFF and extends downwards

When entering a debug session and inspecting sbss and ebss i find that the .bss section is outside ram.
[image]

Both the memory.x and [link.x](https://github.com/japaric/cortex-m-rt/blob/master/link.x) seems sane so not exactly sure where this bug is. 

Im also quite certain that the memory.x script is in fact linked in as the stack pointer seem to start at 0x1FFFFFFF.

pftbest, 2017-05-28:
Have you tried to build `--release`?

japaric, 2017-05-28:
I'm as puzzled as you. Maybe the register got swapped somehow in the zero_bss call? Could you check that the bss section starts at 0x1FFF1000 with `arm-none-eabi-size -Ax $binary`, and that the _ebss and _sbss symbols have the right values with `arm-none-eabi-nm -C $binary`?

kjetilkjeka, 2017-05-28:
@pftbest the behaviour is analog when compiling with --release. The only difference is that i can't read out n since it's optimized away.

This won't happen when I compile in debug as I'm currently using
    [profile.dev]
    opt-level = 0

kjetilkjeka, 2017-05-28:
@japaric finally something that looks somewhat sane

arm-none-eabi-size -Ax target/thumbv7em-none-eabihf/debug/examples/light
target/thumbv7em-none-eabihf/debug/examples/light  gives
section                  size     addr
    .vector_table           0x400      0x0
    .text                   0xc24    0x400
    .rodata                 0x1bc   0x1030
    .debug_gdb_scripts       0x22      0x0
    .debug_str            0x4ae07      0x0
    .debug_loc            0x31cde      0x0
    .debug_abbrev           0xb89      0x0
    .debug_info          0x18dddf      0x0
    .debug_ranges         0x19408      0x0
    .debug_macinfo            0x2      0x0
    .debug_pubnames        0x81ba      0x0
    .debug_pubtypes       0x2f1d8      0x0
    .ARM.attributes          0x36      0x0
    .debug_frame          0x22a1c      0x0
    .debug_line           0x529d0      0x0
    Total                0x2d220d

arm-none-eabi-nm -C target/thumbv7em-none-eabihf/debug/examples/light | grep _ebss
    1fff1000 N _ebss

and
arm-none-eabi-nm -C target/thumbv7em-none-eabihf/debug/examples/light | grep _sbss
    1fff1000 N _sbss

I assume it's not listed for size since the size is 0.

I guess the next thing to find out is what happened when doing the zero_bss call

japaric, 2017-05-28:
Hold on

> thumbv7em-none-eabihf

That's the target for Cortex-M4F processors. I thought your board had a Cortex-M0+ processor. You should be using the thumbv6m-none-eabi target.

kjetilkjeka, 2017-05-28:
The board infact has a Cortex-M4F processor. The lower tier in the S32 series uses a M0+ processor while the upper tier uses a M4F processor. 

The board I'm using is [S32K144EVB](http://www.nxp.com/products/automotive-products/microcontrollers-and-processors/arm-mcus-and-mpus/s32-processors-microcontrollers/s32k144-evaluation-board:S32K144EVB) You can confirm the processor core on page 5 in the [reference manual ](http://www.nxp.com/assets/documents/data/en/data-sheets/S32K-DS.pdf)

kjetilkjeka, 2017-05-28:
It can see from the assembly pane in the debugging session that _bbss and _ebss is loaded as immediate (of course as expected). Which i guess means that there is some fish linking going on? Any idea how that is possible?

kjetilkjeka, 2017-05-28:
Since the dissassembly gained with:

    arm-none-eabi-objdump -d target/thumbv7em-none-eabihf/debug/examples/light

Shows sane code it's not the cortex-m-rt or the linkers fault.

I guess this issue can be closed. I will redirect my frustration to the kind people at Segger. Thank you for rapid reply and assistance.

[image]

pftbest, 2017-05-28:
@kjetilkjeka Can you please give me your binary? I'd like to play with it a little in qemu

kjetilkjeka, 2017-05-28:
Of course, here it is:
[light.zip](https://github.com/japaric/cortex-m-rt/files/1034293/light.zip)

pftbest, 2017-05-28:
Yeah, this binary works fine in qemu, so you are probably right, there might be something wrong with your debugger.

kjetilkjeka, 2017-05-29:
This bug was due to configuration bytes placed where .text is usually placed in the S32K MCU. Meaning cortex-m-rt places .bss just where it should.

See #13 for the actual issue.

closed 2017-05-29 by kjetilkjeka