rust-embedded/cortex-m-rt #15: Compiling fails due to `compiler_builtins`

nixpulvis, 2017-06-05:
```
error[E0463]: can't find crate for `compiler_builtins`
   --> /Users/nixpulvis/.cargo/git/checkouts/cortex-m-rt-c229622fecd03225/d11471e/src/lib.rs:160:1
    |
160 | extern crate compiler_builtins;
    | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ can't find crate
```

I've even tried adding it as a dev to my crate, with no luck. Here's my `Cargo.toml`.

```toml
[package]
name = "rtfm-playground"
version = "0.1.0"

[dependencies]
compiler_builtins = { git = "https://github.com/rust-lang-nursery/compiler-builtins" }
cortex-m = { git = "https://github.com/japaric/cortex-m.git" }
cortex-m-rt = { git = "https://github.com/japaric/cortex-m-rt.git" }
cortex-m-rtfm = { git = "https://github.com/japaric/cortex-m-rtfm.git" }
```

japaric, 2017-06-05:
> I've even tried adding it as a dependency to my crate

Adding it as a dependency of the *top* crate is not going to work. It has to be in the Xargo sysroot. Are you using a Xargo.toml file as [the documentation] indicates?

[the documentation]: https://docs.rs/cortex-m-rt/0.2.4/cortex_m_rt/#example

Also you should be using the [quickstart template] to avoid this and others problems. Its documentation even has an extensive [troubleshooting] section. If you want a more extensive step by step guide of how to use the quickstart template check [this blog post].

[quickstart template]: https://docs.rs/cortex-m-quickstart/0.1.8/cortex_m_quickstart/
[troubleshooting]: https://docs.rs/cortex-m-quickstart/0.1.8/cortex_m_quickstart/#troubleshooting
[this blog post]: http://blog.japaric.io

japaric, 2018-01-17:
@nixpulvis I'm going to assume things worked out since you didn't reply after my last comment :-). If you run into any other problem feel free to open a new issue!

closed 2018-01-17 by japaric