rust-embedded/cortex-m-rt #17: add some unsafe method to run code before RAM is initialized

japaric, 2017-06-14:
Some people have expressed that they would like to configure the CPU clock before RAM initialization to speed up the initialization process (specially when lots of RAM need to be initialized).

This has to be unsafe because all the static variables stored in .bss / .data would be uninitialized during the execution of this pre-init routine.

Possible user facing interface:

``` rust
#[macro_use]
extern crate cortex_m_rt;

pre_init!(unsafe {
    // something
});
```

where `pre_init!` can only be used once.

djmcgill, 2017-07-24:
How do you recommend doing this? I was thinking of either something like https://github.com/rust-lang/rfcs/issues/1971 i.e. a default trait implementation and have the macro override it, or adding a new (weakly linked) function and then overriding that a la the r0 crate's init_array.

japaric, 2017-08-01:
The latter, like we do for interrupt / exception handlers.

jcsoo, 2018-04-26:
Another use case is disabling watchdogs before main() on some devices such as the K64 that have them enabled on reset. Unfortunately, leaving the watchdog enabled can leave the device in a reset loop that is difficult to recover from.

I would also like to enable instruction cache on devices that support it, and enter main() with interrupts disabled by default.

It sounds like the right way to do this is to add a weakly linked pre_init function that
is NOP by default but that can be overridden, and then call that within reset_handler().

Or would it be better to have a way to replace reset_handler?

thejpster, 2018-06-01:
I needed this on the AM5728 IPU1 as both Cortex-M4 cores (there are two) are released out of reset at the same time with the same reset vector. You write a little bit of code (making sure not to use the stack pointer) to put Core 1 to sleep and let Core 0 carry on. You can then take Core 1 out of sleep at a later time if you're running an SMP-capable RTOS.

Currently I fixed this by forking the crate.

commit c5349326 (2018-08-13): Merge #81

81: Add a __pre_init function to be called at the start of the reset handler r=korken89 a=yodaldevoid

I needed this for a project so I went ahead and rebased and tweaked @jcsoo 's changes from #71. I will admit, I got a little impatient waiting and that also played into why I did this. If either @jcsoo or @japaric have an issue with this PR, please let me know and I will remove it. I apologize for toes that I have stepped on.

Now onto the PR. This is possibly the simplest implementation that is possible. No nightly features were used in this implementation. Also, there is no hand- [... cut]
PR #81 (2018-08-13): Add a __pre_init function to be called at the start of the reset handler
I needed this for a project so I went ahead and rebased and tweaked @jcsoo 's changes from #71. I will admit, I got a little impatient waiting and that also played into why I did this. If either @jcsoo or @japaric have an issue with this PR, please let me know and I will remove it. I apologize for toes that I have stepped on.

Now onto the PR. This is possibly the simplest implementation that is possible. No nightly features were used in this implementation. Also, there is no hand-holding for the end user; if they want to [... cut]
closed 2018-08-13 by bors[bot]