rust-embedded/cortex-m-rt #22: Global offset table not initialized by bootstrap code.

jannic, 2017-06-30:
When linking a C library compiled as position independent code, the resulting binary contains a section `.got` which contains a 'global offset table'. This section needs to be initialized like `.data` by initialization code, as far as I can tell.

Symptoms were accesses to random memory addresses at runtime, of course leading to a non-working binary.

I found two changes I could do to avoid this issue:
- Either not compile the C library as PIC code. In my case, this was possible using [gcc::Config::pic](https://docs.rs/gcc/0.3.51/gcc/struct.Config.html#method.pic), which defaults to `true`.
- Or add `.got` to the `.data` section using a modified link script:
```
diff --git a/link.x b/link.x
index e6c90fa..f820b73 100644
--- a/link.x
+++ b/link.x
@@ -43,7 +43,7 @@ SECTIONS
   .data : ALIGN(4)
   {
     _sdata = .;
-    *(.data .data.*);
+    *(.data .data.* .got);
     _edata = ALIGN(4);
   } > RAM AT > FLASH
```

Each one of those changes lead to a working binary. But I'm not sure if changing the link script is a proper solution for this issue or just a dirty workaround.
Maybe PIC code doesn't even make much sense for typical cortex-m use cases. But currently, this leads to hard to debug issues, as there isn't even a warning, just a broken binary.
So, just generating an error message when one tries to link PIC code would help, as well. But I have no idea how to achieve that.

japaric, 2017-06-30:
All the thumb targets have "relocation-model" set to "static" in their specification, instead of "pic" which is common for targets with an OS; those OS targets also have the "position-independent-executables" field set to true in their specification whereas the thumb targets set that to false. So, at least, the Rust side won't generate any GOT or relocatable code.

I'd say it's better to compile the C code without PIC. AFAICT, the only thing that compiling with PIC will buy you is that you would be able to dynamically load C code, and only C code because the Rust code is not relocatable -- unless you are using a custom target instead of the built-in thumb ones (Tock OS does this; it uses custom targets for their userspace Rust programs).

> So, just generating an error message when one tries to link PIC code would help, as well.

Maybe something like this (untested):

```
SECTIONS {
  /* .. */
  .got :
  {
    _sgot = .;
    *(.got .got.*);
    _egot = .;
  }
}

ASSERT(_egot - _sgot == 0, "
.GOT section detected. Recompile your C code without -fPIC.
(cf. gcc::Config::pic)");
```

smbolton, 2017-07-18:
Ouch, I got bit by this as well. +1 for changing the linker script to abort if a .got is present.

Thanks for all your awesome work on embedded Rust, Jorge!

japaric, 2017-07-18:
@jannic @smbolton Any of you has a repro case? Do I just have to build some C code using `gcc`? Would any C code make do?

smbolton, 2017-07-18:
Give this a try: https://github.com/smbolton/stm32f3-got-mcve
Build, flash to F3, run, and it should HardFault in c_test_function() because the address it's using to access data[] is wrong.

japaric, 2017-07-18:
@smbolton Thanks for the repro case. #28 raises an error if relocatable code is found in one of the input files.

closed 2017-07-18 by japaric

jannic, 2017-07-19:
Thanks, works great!
(And sorry, didn't have time to upload my test case, yesterday. Glad that smbolton stepped in!)

smbolton, 2017-07-19:
Thanks! (I can't test easily now, as the stm32f30x crate is pinned to cortex-m v0.2.x, which leads to version conflicts, but I'm sure this will save me time down the road, if I ever forget the gcc::Config::pic(false)!)

PR #28 (2017-07-18): raise an error if a .got section is found in the input
the thumb targets don't support dynamic relocations by default and this crate
doesn't include the required startup code either so just error if the user is
attempting to include relocatable code.

closes #22
commit 7b5eb7f2 (2017-07-18): raise an error if a .got section is found in the input

the thumb targets don't support dynamic relocations by default and this crate
doesn't include the required startup code either so just error if the user is
attempting to include relocatable code.

closes #22