rust-embedded/cortex-m-rt #26: Static byte resources in .rodata are not properly aligned/padded

therealprof, 2017-07-16:
Static resources which end up in .rodata are not properly aligned which is especially evident with strings. I've defined a string in my executable like
```
    let s = &"Hello World!\n";
```
which resulted in an invalid binary. Most tools will complain about such a binary and of course there's no way this will run; e.g. st-flash will say something like:
```
2017-07-16T15:31:07 INFO src/common.c: Device connected is: F04x device, id 0x10006445
2017-07-16T15:31:07 INFO src/common.c: SRAM size: 0x1800 bytes (6 KiB), Flash: 0x8000 bytes (32 KiB) in pages of 1024 bytes
2017-07-16T15:31:07 INFO src/common.c: Attempting to write 1025 (0x401) bytes to stm32 address: 134217728 (0x8000000)
2017-07-16T15:31:07 WARN src/common.c: unaligned len 0x401 -- padding with zero
Flash page at addr: 0x08000400 erased
2017-07-16T15:31:07 INFO src/common.c: Finished erasing 2 pages of 1024 (0x400) bytes
```
A black magic probe will even hang completely after this:
```
(gdb) load
Loading section .vector_table, size 0xc0 lma 0x8000000
Loading section .text, size 0x334 lma 0x80000c0
Loading section .rodata, size 0xd lma 0x80003f4
Error writing data to flash
```

therealprof, 2017-07-16:
This patch fixes the problem:
```
diff --git a/link.x b/link.x
index f7f83cb..387edb7 100644
--- a/link.x
+++ b/link.x
@@ -38,6 +38,7 @@ SECTIONS
   .rodata : ALIGN(4)
   {
     *(.rodata .rodata.*);
+    . = ALIGN(4);
   } > FLASH

   .bss : ALIGN(4)
```

japaric, 2017-07-16:
Is this a Cortex-M0 requirement? I don't see any problem with OpenOCD and Cortex M3 and M4 devices. Maybe OpenOCD automatically pads the section that will go in Flash, IDK.

In any case, the patch seems fine to apply.

therealprof, 2017-07-16:
Seems like a general requirement to me; also objdump will output nonsense if it stumbles across a non-padded .rodata. Maybe you just got lucky and never ran into that scenario before?

japaric, 2017-07-16:
> Maybe you just got lucky and never ran into that scenario before?

Nah, just tested flashing and running a program with unaligned .rodata / Flash on a M3 device (used OpenOCD) and it worked fine.

> also objdump will output nonsense if it stumbles across a non-padded .rodata.

I see no cange in the objdump output. What are you observing on your side?

therealprof, 2017-07-16:
The disassembly dissector spits out an "Address is out of bounds":
```
Disassembly of section .rodata:

080003f4 <str.c>:
 80003f4:       6c6c6548        cfstr64vs       mvdx6, [ip], #-288      ; 0xfffffee0
 80003f8:       6f57206f        svcvs   0x0057206f
 80003fc:       21646c72        smccs   18114   ; 0x46c2
 8000400:       Address 0x0000000008000400 is out of bounds.

Disassembly of section .bss:
```

therealprof, 2017-07-16:
Hah! The problem actually only occurs if the .rodata section is followed by a .bss section. For some reason unknown to me I have no idea what triggers the emission of a .bss section, at the moment my code which used to crash before doesn't have one...

therealprof, 2017-07-16:
I openend a ticket for the BlackMagicProbe related to this issue, too: https://github.com/blacksphere/blackmagic/issues/259

japaric, 2017-07-17:
> The problem actually only occurs if the .rodata section is followed by a .bss section. 

.bss, not .data? .bss has no LMA (Load Memory Address) in Flash; how could it affect loading Flash?

I was thinking that maybe the .data section LMA, which goes in Flash, may not always be aligned. The linker script may not be forcing it to be aligned to a 4 byte boundary. If that's the case then the .data section LMA will be unaligned if the end of the .bss is not aligned to a 4 byte boundary.

japaric, 2017-07-17:
> I was thinking that maybe the .data section LMA, which goes in Flash, may not always be aligned

I checked and no, it is 4 byte aligned. I've pushed a new commit that pads .bss and .data so their sizes are always multiple of 4 bytes. This fixes the "Address (..) is out of bounds." `objdump` message you have observed.

therealprof, 2017-07-17:
> .bss, not .data? .bss has no LMA (Load Memory Address) in Flash; how could it affect loading Flash?

I think debuggers will load the .bss section in RAM if there's one in the binary.

japaric, 2017-07-18:
> I think debuggers will load the .bss section in RAM if there's one in the binary

I think I *might* have seen something like that before, but I'm not sure if what I was observing was RAM previously initialized by e.g. r0::{init_data,zero_bss} whose contents were preserved across reboots / debug sessions, or the debugger initializing RAM.

> What is not quite clear to me is why I sometimes have a .bss section and when I tried to reproduce it I didn't.

Hmm, maybe some Rust function contains some static variable in it? If that's the case that may show up in dev mode but not in release mode.

In any case, does using the tip of the master branch solves this problem for you? All the sections should now be padded to have a size multiple of 4 bytes. You can try out the master branch by adding these lines to your Cargo.toml:

``` toml
[replace]
"cortex-m-rt:0.3.3" = { git = "https://github.com/japaric/cortex-m-rt" }
```

You might need to remove / update the Cargo.lock file for this change to work.

therealprof, 2017-07-18:
I've overridden the the link.x with my proposed patch since I've found it but your version works at least as well and fixes the problem, yes.

japaric, 2017-07-18:
Thanks for checking. Closing as fixed then.

closed 2017-07-18 by japaric