rust-embedded/cortex-m-rt #29: Separate default handler for ADC_COMP emitted

therealprof, 2017-07-22:
For some reason I couldn't quite figure out there's still a separate default handler for ADC_COMP emitted:

```
080002c4 <_ZN11cortex_m_rt15default_handler17h264a3b21ccda1ce5E>:
 80002c4:       be00            bkpt    0x0000
 80002c6:       e7fe            b.n     80002c6 <_ZN11cortex_m_rt15default_handler17h264a3b21ccda1ce5E+0x2>

080002c8 <ADC_COMP>:
 80002c8:       f3ef 8008       mrs     r0, MSP
 80002cc:       e7fa            b.n     80002c4 <_ZN11cortex_m_rt15default_handler17h264a3b21ccda1ce5E>
```

therealprof, 2017-07-22:
DOH! How could I possibly miss that, you're absolutely right.

adamgreig, 2018-08-12:
@therealprof are you happy to close this issue?

therealprof, 2018-08-12:
I totally forgot about this. Sure.

closed 2018-08-12 by therealprof