rust-embedded/cortex-m-rt #31: remove dependency on `compiler_builtins`

japaric, 2017-07-23:
That crate is an implementation detail of the compiler. It contains the implementation of intrinsics that LLVM may emit calls to when lowering Rust code to machine code. It's unknown what the future of `compiler_builtins` will be but likely it will not exist in the future (it *might* be merged into the core crate).

So this a breaking change that will come from upstream at some point: `extern crate compiler_builtins` is unstable.

djmcgill, 2017-07-24:
How will this change be effected? Just delete all references to `compiler_builtins` and then the user must include and link it themselves to use this crate?

I'm very happy to start work on this.

japaric, 2017-07-31:
I think we should not change things right now. Because if we do we are going to break code *and* incovenience all application writers (they'll now have to manually link to compiler-builtins), and *then* whenever the Rust team makes a decision about compiler-builtins things are likely going to break *again*.

I would prefer to maintain the status quo until an official decision is made about compiler-builtins.

commit b6c8fd65 (2018-04-08): drop compiler-builtins bits

they were leftover from rebasing

closes #31
closed 2018-04-09 by japaric