rust-embedded/cortex-m-rt #33: Linker warning recommends a nonexistant `gcc::Config.fpic` method

Wallacoloo, 2017-09-20:
When using bindgen + gcc_rs to wrap bare metal C code, gcc_rs defaults to building with -fPIC, even though dynamic relocation never happens, making position independent code nonessential.

At link time when linking to a library built with -fPIC, the cortex-m-rt link script will error with:

> .got section detected in the input files. Dynamic relocations are not
> supported. If you are linking to C code compiled using the `gcc` crate
> then modify your build script to compile the C code _without_ the
> -fPIC flag. See the documentation of the `gcc::Config.fpic` method for
> details.

However, `gcc::Config.fpic` doesn't appear anywhere in the [gcc-rs docs](http://alexcrichton.com/gcc-rs/gcc/struct.Config.html). The solution for me was to use the `gcc::Config.flag` method, passing it `-fno-pic`, e.g.

```
gcc::Build::new().flag("-fno-pic")
```

I'm not sure if I'm misunderstanding the warning or if it's outdated. If the latter, I tracked the error message down to [here](https://github.com/japaric/cortex-m-rt/blob/master/link.x#L126) & I'm happy to make a PR that updates the message if requested.

japaric, 2017-09-21:
It seems that `Config` has been renamed to `Build` in v0.3.51; `Config` is still around but it's deprecated. In any case, the message in wrong because the method's name is `fpic`, not `pic`. That should be fixed.

japaric, 2018-01-17:
> the message is wrong because the method's name is fpic, not pic

This has been fixed.

closed 2018-01-17 by japaric