rust-embedded/cortex-m-rt #35: Interrupt handlers error

csagonero, 2017-10-02:
I 'm trying to compile the blinky.rs example for STM32F072RBT6

`xargo rustc --release --target thumbv6m-none-eabi --example blinky -- -C link-arg=-Tlink.x -C linker=arm-none-eabi-ld -Z linker-flavor=ld`

I'm getting this error :
```
arm-none-eabi-ld: 
          Invalid '.vector_table.exceptions' section.
arm-none-eabi-ld: 
          The interrupt handlers are missing. If you are not linking to a device
          crate then you supply the interrupt handlers yourself. Check the
          documentation.
```
May be I missed something ...

japaric, 2017-10-02:
Do you happen to be using a nightly toolchain newer than `1.22.0-nightly (325ba23d5 2017-09-19)`? I'm suspecting that there may have been some regression in recent rustcs. Someone reported a similar problem in japaric/cortex-m-quickstart#17. I suggest trying an older toolchain release.

closed 2017-10-03 by japaric

japaric, 2017-10-03:
Fixed in cortex-m-rt v0.3.6. Affected users try `cargo update`.