rust-embedded/cortex-m-rt #39: Do not use the global_asm feature

whitequark, 2017-10-15:
As discussed, see https://github.com/japaric/cortex-m/issues/19#issuecomment-324171957 for details.

whitequark, 2017-10-15:
The patches are in LLVM as r315852, r315853.

whitequark, 2017-10-15:
Can someone pull in https://github.com/llvm-mirror/llvm/commit/e53e8bdb9baba4e9f5ae32c50c04184780c07a0b and https://github.com/llvm-mirror/llvm/commit/99dc03a8d7128708d11433b72b549583c9c02e50 into https://github.com/rust-lang/llvm? I'm not sure what the process is.

whitequark, 2017-10-15:
You might also need https://github.com/llvm-mirror/llvm/commit/f9f40b539b1dc44b10184684aa3f72e1192c454f depending on where that repo has branched from.

japaric, 2017-11-22:
@whitequark I tried your LLVM changes (the 3 patches) locally but I couldn't get the function deduplication to work.

I replaced `global_asm!` with plain function calls for the core exceptions as seen in [this
branch] but I still see function duplication:

[this branch]: https://github.com/japaric/cortex-m-rt/tree/gh39

``` console
08000008 <EXCEPTIONS>:
 8000008:       08000487        stmdaeq r0, {r0, r1, r2, r7, sl}
 800000c:       0800048d        stmdaeq r0, {r0, r2, r3, r7, sl}
 8000010:       08000493        stmdaeq r0, {r0, r1, r4, r7, sl}
 8000014:       08000499        stmdaeq r0, {r0, r3, r4, r7, sl}
 8000018:       0800049f        stmdaeq r0, {r0, r1, r2, r3, r4, r7, sl}
        ...
 800002c:       080004a5        stmdaeq r0, {r0, r2, r5, r7, sl}
        ...
 8000038:       080004ab        stmdaeq r0, {r0, r1, r3, r5, r7, sl}
 800003c:       080004b1        stmdaeq r0, {r0, r4, r5, r7, sl}

(..)

08000486 <NMI>:
 8000486:       f000 f816       bl      80004b6 <DEFAULT_HANDLER>
 800048a:       defe            udf     #254    ; 0xfe

0800048c <HARD_FAULT>:
 800048c:       f000 f813       bl      80004b6 <DEFAULT_HANDLER>
 8000490:       defe            udf     #254    ; 0xfe

08000492 <MEM_MANAGE>:
 8000492:       f000 f810       bl      80004b6 <DEFAULT_HANDLER>
 8000496:       defe            udf     #254    ; 0xfe

08000498 <BUS_FAULT>:
 8000498:       f000 f80d       bl      80004b6 <DEFAULT_HANDLER>
 800049c:       defe            udf     #254    ; 0xfe

0800049e <USAGE_FAULT>:
 800049e:       f000 f80a       bl      80004b6 <DEFAULT_HANDLER>
 80004a2:       defe            udf     #254    ; 0xfe

080004a4 <SVCALL>:
 80004a4:       f000 f807       bl      80004b6 <DEFAULT_HANDLER>
 80004a8:       defe            udf     #254    ; 0xfe

080004aa <PENDSV>:
 80004aa:       f000 f804       bl      80004b6 <DEFAULT_HANDLER>
 80004ae:       defe            udf     #254    ; 0xfe

080004b0 <SYS_TICK>:
 80004b0:       f000 f801       bl      80004b6 <DEFAULT_HANDLER>
 80004b4:       defe            udf     #254    ; 0xfe

080004b6 <DEFAULT_HANDLER>:
 80004b6:       f3ef 8008       mrs     r0, MSP
 80004ba:       f000 b801       b.w     80004c0 <cortex_m_rt::default_handler>
 80004be:       defe            udf     #254    ; 0xfe

080004c0 <cortex_m_rt::default_handler>:
 80004c0:       be00            bkpt    0x0000
 80004c2:       e7fe            b.n     80004c2 <cortex_m_rt::default_handler+0x2>
```

Passing `-C passes=mergefunc` to `rustc` didn't help. I also tried `asm!("b DEFAULT_HANDLER")`
instead of a function call but the result was the same.

It's been a while since I looked into this so I may be doing something wrong.

P.S. I could be misremembering but I don't recall the exception handlers including an abort
instruction (`udf 0xfe`) in their bodies; they were just a single branch (`b`) instruction, I think.

whitequark, 2017-11-23:
That's weird. Care to give me the `.ll`?

japaric, 2017-11-24:
Ohh, I must have been doing something wrong when testing with `-C passes=mergefunc` because I'm now getting an LLVM error when compiling the compiler-builtins crate or the *patched* cortex-m-rt (v0.3.6 compiles fine with mergefunc).

``` console
$ cat .cargo/config
[target.thumbv7m-none-eabi]
runner = 'arm-none-eabi-gdb'
rustflags = [
  "-C", "link-arg=-Tlink.x",
  "-C", "linker=arm-none-eabi-ld",
  "-C", "passes=mergefunc", # <-
  "-Z", "linker-flavor=ld",
]

[build]
target = "thumbv7m-none-eabi"

$ rustup default nightly

$ xargo build --release && echo OK
OK

$ rustup default stage1 # with your LLVM patches

$ cargo clean; xargo build --release -v
+ "rustc" "--print" "sysroot"
+ "rustc" "--print" "target-list"
+ "cargo" "build" "--release" "--manifest-path" "/tmp/xargo.zcyfvDsh5p8w/Cargo.toml" "--target" "thumbv7m-none-eabi" "-v" "-p" "core"
   Compiling core v0.0.0 (file:///home/japaric/rust/lang/src/libcore)
     Running `rustc --crate-name core /home/japaric/rust/lang/src/libcore/lib.rs --crate-type lib --emit=dep-info,link -C opt-level=3 -C debuginfo=2 -C metadata=93d0fd8cf485fc37 -C extra-filename=-93d0fd8cf485fc37 --out-dir /tmp/xargo.zcyfvDsh5p8w/target/thumbv7m-none-eabi/release/deps --target thumbv7m-none-eabi -L dependency=/tmp/xargo.zcyfvDsh5p8w/target/thumbv7m-none-eabi/release/deps -L dependency=/tmp/xargo.zcyfvDsh5p8w/target/release/deps -C link-arg=-Tlink.x -C linker=arm-none-eabi-ld -C passes=mergefunc -Z linker-flavor=ld --sysroot /home/japaric/.xargo -Z force-unstable-if-unmarked`
    Finished release [optimized + debuginfo] target(s) in 12.22 secs
+ "cargo" "build" "--release" "--manifest-path" "/tmp/xargo.7ifZ0QMksUzT/Cargo.toml" "--target" "thumbv7m-none-eabi" "-v" "-p" "compiler_builtins"
   Compiling compiler_builtins v0.1.0 (file:///home/japaric/rust/lang/src/libcompiler_builtins)
     Running `rustc --crate-name build_script_build /home/japaric/rust/lang/src/libcompiler_builtins/build.rs --crate-type bin --emit=dep-info,link -C opt-level=3 -C debuginfo=2 --cfg 'feature="compiler-builtins"' --cfg 'feature="default"' --cfg 'feature="mem"' -C metadata=91f17aa6bbbd60f8 -C extra-filename=-91f17aa6bbbd60f8 --out-dir /tmp/xargo.7ifZ0QMksUzT/target/release/build/compiler_builtins-91f17aa6bbbd60f8 -L dependency=/tmp/xargo.7ifZ0QMksUzT/target/release/deps`
     Running `/tmp/xargo.7ifZ0QMksUzT/target/release/build/compiler_builtins-91f17aa6bbbd60f8/build-script-build`
     Running `rustc --crate-name compiler_builtins /home/japaric/rust/lang/src/libcompiler_builtins/src/lib.rs --crate-type lib --emit=dep-info,link -C opt-level=3 -C debuginfo=2 --cfg 'feature="compiler-builtins"' --cfg 'feature="default"' --cfg 'feature="mem"' -C metadata=8501aab9e9d5f9f2 -C extra-filename=-8501aab9e9d5f9f2 --out-dir /tmp/xargo.7ifZ0QMksUzT/target/thumbv7m-none-eabi/release/deps --target thumbv7m-none-eabi -L dependency=/tmp/xargo.7ifZ0QMksUzT/target/thumbv7m-none-eabi/release/deps -L dependency=/tmp/xargo.7ifZ0QMksUzT/target/release/deps -C link-arg=-Tlink.x -C linker=arm-none-eabi-ld -C passes=mergefunc -Z linker-flavor=ld --sysroot /home/japaric/.xargo -Z force-unstable-if-unmarked --cfg thumb`
LLVM ERROR: Unexpected anonymous function when writing summary
```

LLVM assertions are now disabled by default in rust-lang/rust as of last week so my build also has them disabled. Perhaps this would have produced a more helpful message with assertions enabled?

Here are [the .ll files](https://gist.github.com/japaric/76ba49051540dee4f3cf81db9efe2ddd) *without* `-C passes=mergfunc`. They look pretty much the same except for the differences in hashes / mangling so probably `mergefunc` is disabled in the `--release` passes.

japaric, 2018-05-13:
We are no longer using `global_asm!` (or any other unstable feature). Weak aliasing (of exception / interrupt handlers to the `DefaultHandler`) is implemented using linker scripts.

closed 2018-05-13 by japaric