rust-embedded/cortex-m-rt #44: Exception configuration support?

therealprof, 2017-11-13:
It seems there's currently no good way to configure system exceptions at the moment. Neither does svd2rust generate a code for the configuration of the SCB nor does cortex-m-rt provide provide proper support for all registers.

Without the ability to configure the SCB it is e.g. not possible to change the priorities of user configurable exceptions like SysTick which means it'll always run at the highest priority thus pre-empting or at least competing (if interrupts are all set to highest priority, too) with all other interrupts when used.

therealprof, 2017-11-14:
Meh, of course this was meant for *cortex-m*, but the point still stands.

japaric, 2017-11-14:
cortex-m exposes an API for [SCB](https://docs.rs/cortex-m/0.3.1/cortex_m/peripheral/struct.SCB.html). cortex-m-rtfm [uses] that API (plain writes to `shpr`) to set the priority of user exceptions. If you would like to suggest or implement a higher level API around that (like we have for NVIC) please open an issue / PR in the cortex-m repo.

[uses]: https://github.com/japaric/cortex-m-rtfm/blob/c184f91e3c3993dacbd8425019cde6a3607478bc/macros/src/trans.rs#L280

I'm closing this issue since it's in the wrong repository and because a low level mechanism for configuring the priority of user exceptions already exists.

closed 2017-11-14 by japaric

therealprof, 2017-11-15:
Yes, I'll definitely do that. The ergonomics of the "low-level" interface is so horrible (yet still unsafe) that I ended up doing a simple `unsafe { ptr::write_volatile(0xE000_ED20 as *mut u32, 0xC000_0000) }` instead of using it.