rust-embedded/cortex-m-rt #47: Missing DebugMonitor Exception

yodaldevoid, 2017-12-21:
According to the ARMv7-M Architecture Reference Manual there should be a DebugMonitor exception handler after the one for SVCall. This crate does not expose this exception. Was this done intentionally or was the DebugMonitor exception simply overlooked?

japaric, 2017-12-21:
The vector table figure from the official ARM documentation indicates that the slot you mention is "Reserved for Debug". This was probably understood as *don't put any user application handler here*, which is the case for all the other "Reserved" spots in the table.

[image]

Do you have a link to official documentation that indicates that this is a user configurable exception handler (and that mentions the use case)? To me "Reserved for Debug" sounds like this is either used by something like semihosting or some debugging solution (e.g. Segger J-Link).

yodaldevoid, 2017-12-21:
This is indeed used for the Segger J-Link and other debuggers, but requires code which acts as the handler for this exception. Looking at the [sample project](https://www.segger.com/products/debug-probes/j-link/technology/monitor-mode-debugging/) using the monitor mode debugging gives a sample case. Writing code compatible with any debug solution relying on the DebugMonitor exception is impossible without a way to set the exception handler.

As for official documentation from ARM, I will admit I could not find anything specifically stating the purpose of the DebugMonitor exception. the closest I could find was In the [ARMv7-M Architecture Reference Manual](https://static.docs.arm.com/ddi0403/eb/DDI0403E_B_armv7m_arm.pdf) page 754 talks about code stepping using the debug monitor and talks about a DebugMonitor handler. In addition table B1-4 on page 581 lists vector 12 as DebugMonitor, not Reserved for Debug, and the wording of the description of the exception, found on page 580, cause me to believe that it is intended to be user configurable.
[image]

japaric, 2017-12-23:
Thanks for digging this info, @yodaldevoid.

It seems the exception is not available on Cortex-M0(+) so it needs to be provided conditionally.

closed 2017-12-23 by japaric

yodaldevoid, 2017-12-23:
Thank you for your quick work! I can't say I was expecting this to be taken care of so quickly.

japaric, 2017-12-23:
v0.3.7 has been published and it includes #48

PR #48 (2017-12-23): provide a DEBUG_MONITOR exception on ARMv7-M
closes #47
cc @yodaldevoid