rust-embedded/cortex-m-rt #49: arm-none-eabi-ld not working with v0.3.8

clebi, 2017-12-29:
Hello,

I have updated to version 0.3.8 of this crate to avoid this issue : [ICE when trying to cross compile with xargo](https://github.com/rust-lang/rust/issues/47040).
It now compiles but it seems there is a problem when linking.

I am using the cortex-m-rtfm framework and stm32f3disovery board. I use the f3 crate (latest git) and I was successful compiling the exact same program before.

It seems the problem is coming from interrupts. I have the following interrupts setup : I2C1_EV_EXTI23 and I2C1_ER. When I remove all interrupts, program compiles and links successfully. 

Here is the nightly I am using : rustc 1.24.0-nightly (77e189cd7 2017-12-28)

This is the ld output : 

```
   Compiling i2c_magnetic v0.1.0 (file:///home/clement/atom-projects/cortex/i2c_magnetic)
error: linking with `arm-none-eabi-ld` failed: exit code: 1
  |
  = note: "arm-none-eabi-ld" "-L" "/home/clement/.xargo/lib/rustlib/thumbv7em-none-eabihf/lib" "/home/clement/atom-projects/cortex/i2c_magnetic/target/thumbv7em-none-eabihf/debug/deps/i2c_magnetic-c094ee808221889b.__rustc_fallback_codegen_unit.rcgu.o" "/home/clement/atom-projects/cortex/i2c_magnetic/target/thumbv7em-none-eabihf/debug/deps/i2c_magnetic-c094ee808221889b.bare_metal.volatile.rcgu.o" "/home/clement/atom-projects/cortex/i2c_magnetic/target/thumbv7em-none-eabihf/debug/deps/i2c_magnetic-c094ee808221889b.core-cell.volatile.rcgu.o" "/home/clement/atom-projects/cortex/i2c_magnetic/target/thumbv7em-none-eabihf/debug/deps/i2c_magnetic-c094ee808221889b.core-char.rcgu.o" "/home/clement/atom-projects/cortex/i2c_magnetic/target/thumbv7em-none-eabihf/debug/deps/i2c_magnetic-c094ee808221889b.core-fmt.rcgu.o" "/home/clement/atom-projects/cortex/i2c_magnetic/target/thumbv7em-none-eabihf/debug/deps/i2c_magnetic-c094ee808221889b.core-fmt.volatile.rcgu.o" "/home/clement/atom-projects/cortex/i2c_magnetic/target/thumbv7em-none-eabihf/debug/deps/i2c_magnetic-c094ee808221889b.core-num.rcgu.o" "/home/clement/atom-projects/cortex/i2c_magnetic/target/thumbv7em-none-eabihf/debug/deps/i2c_magnetic-c094ee808221889b.core-ops-range.volatile.rcgu.o" "/home/clement/atom-projects/cortex/i2c_magnetic/target/thumbv7em-none-eabihf/debug/deps/i2c_magnetic-c094ee808221889b.core-ptr.volatile.rcgu.o" "/home/clement/atom-projects/cortex/i2c_magnetic/target/thumbv7em-none-eabihf/debug/deps/i2c_magnetic-c094ee808221889b.core-result.volatile.rcgu.o" "/home/clement/atom-projects/cortex/i2c_magnetic/target/thumbv7em-none-eabihf/debug/deps/i2c_magnetic-c094ee808221889b.core-slice.volatile.rcgu.o" "/home/clement/atom-projects/cortex/i2c_magnetic/target/thumbv7em-none-eabihf/debug/deps/i2c_magnetic-c094ee808221889b.core-str.rcgu.o" "/home/clement/atom-projects/cortex/i2c_magnetic/target/thumbv7em-none-eabihf/debug/deps/i2c_magnetic-c094ee808221889b.cortex_m-peripheral.volatile.rcgu.o" "/home/clement/atom-projects/cortex/i2c_magnetic/target/thumbv7em-none-eabihf/debug/deps/i2c_magnetic-c094ee808221889b.cortex_m_rt-lang_items.volatile.rcgu.o" "/home/clement/atom-projects/cortex/i2c_magnetic/target/thumbv7em-none-eabihf/debug/deps/i2c_magnetic-c094ee808221889b.cortex_m_rtfm.volatile.rcgu.o" "/home/clement/atom-projects/cortex/i2c_magnetic/target/thumbv7em-none-eabihf/debug/deps/i2c_magnetic-c094ee808221889b.cortex_m_semihosting-hio.volatile.rcgu.o" "/home/clement/atom-projects/cortex/i2c_magnetic/target/thumbv7em-none-eabihf/debug/deps/i2c_magnetic-c094ee808221889b.i2c_magnetic.rcgu.o" "/home/clement/atom-projects/cortex/i2c_magnetic/target/thumbv7em-none-eabihf/debug/deps/i2c_magnetic-c094ee808221889b.i2c_magnetic-I2C1_ER.rcgu.o" "/home/clement/atom-projects/cortex/i2c_magnetic/target/thumbv7em-none-eabihf/debug/deps/i2c_magnetic-c094ee808221889b.i2c_magnetic-I2C1_EV_EXTI23.rcgu.o" "/home/clement/atom-projects/cortex/i2c_magnetic/target/thumbv7em-none-eabihf/debug/deps/i2c_magnetic-c094ee808221889b.i2c_magnetic.volatile.rcgu.o" "/home/clement/atom-projects/cortex/i2c_magnetic/target/thumbv7em-none-eabihf/debug/deps/i2c_magnetic-c094ee808221889b.static_ref.volatile.rcgu.o" "-o" "/home/clement/atom-projects/cortex/i2c_magnetic/target/thumbv7em-none-eabihf/debug/deps/i2c_magnetic-c094ee808221889b" "--gc-sections" "-L" "/home/clement/atom-projects/cortex/i2c_magnetic/target/thumbv7em-none-eabihf/debug/deps" "-L" "/home/clement/atom-projects/cortex/i2c_magnetic/target/debug/deps" "-L" "/home/clement/atom-projects/cortex/i2c_magnetic/target/thumbv7em-none-eabihf/debug/build/cortex-m-rt-986ccafb8ce49ac6/out" "-L" "/home/clement/atom-projects/cortex/i2c_magnetic/target/thumbv7em-none-eabihf/debug/build/f3-0569b571bc6c2bfd/out" "-L" "/home/clement/.xargo/lib/rustlib/thumbv7em-none-eabihf/lib" "-Bstatic" "/home/clement/atom-projects/cortex/i2c_magnetic/target/thumbv7em-none-eabihf/debug/deps/libcortex_m_rtfm-6e6f4a688986a923.rlib" "/home/clement/atom-projects/cortex/i2c_magnetic/target/thumbv7em-none-eabihf/debug/deps/librtfm_core-2047840c54fc7908.rlib" "/home/clement/atom-projects/cortex/i2c_magnetic/target/thumbv7em-none-eabihf/debug/deps/libuntagged_option-983345c6d1de8ec5.rlib" "/home/clement/atom-projects/cortex/i2c_magnetic/target/thumbv7em-none-eabihf/debug/deps/libf3-df303aac69aa2c04.rlib" "/home/clement/atom-projects/cortex/i2c_magnetic/target/thumbv7em-none-eabihf/debug/deps/libstm32f30x-3f7de102ea6e6017.rlib" "/home/clement/atom-projects/cortex/i2c_magnetic/target/thumbv7em-none-eabihf/debug/deps/libcortex_m_rt-b675d40d94fb56f4.rlib" "/home/clement/atom-projects/cortex/i2c_magnetic/target/thumbv7em-none-eabihf/debug/deps/libr0-2f265f8ff98e4ae6.rlib" "/home/clement/atom-projects/cortex/i2c_magnetic/target/thumbv7em-none-eabihf/debug/deps/libstatic_ref-0d8916cd22b1d3af.rlib" "/home/clement/atom-projects/cortex/i2c_magnetic/target/thumbv7em-none-eabihf/debug/deps/libembedded_hal-1316f468aecd520d.rlib" "/home/clement/atom-projects/cortex/i2c_magnetic/target/thumbv7em-none-eabihf/debug/deps/libcast-53ac32fcf11bdfe7.rlib" "/home/clement/atom-projects/cortex/i2c_magnetic/target/thumbv7em-none-eabihf/debug/deps/libnb-5107a94c34bdea80.rlib" "/home/clement/atom-projects/cortex/i2c_magnetic/target/thumbv7em-none-eabihf/debug/deps/libcortex_m-18bde24940ccf3ce.rlib" "/home/clement/atom-projects/cortex/i2c_magnetic/target/thumbv7em-none-eabihf/debug/deps/libbare_metal-d5e58ad0a5570173.rlib" "/home/clement/atom-projects/cortex/i2c_magnetic/target/thumbv7em-none-eabihf/debug/deps/libaligned-1ac77a68d10d4b67.rlib" "/home/clement/atom-projects/cortex/i2c_magnetic/target/thumbv7em-none-eabihf/debug/deps/libvolatile_register-fb080589ac340895.rlib" "/home/clement/atom-projects/cortex/i2c_magnetic/target/thumbv7em-none-eabihf/debug/deps/libvcell-95a1d9c5f049d0fb.rlib" "/home/clement/atom-projects/cortex/i2c_magnetic/target/thumbv7em-none-eabihf/debug/deps/libcortex_m_semihosting-953ef35a657c1fc9.rlib" "/home/clement/.xargo/lib/rustlib/thumbv7em-none-eabihf/lib/libcore-12794acabb4ff5e4.rlib" "/home/clement/.xargo/lib/rustlib/thumbv7em-none-eabihf/lib/libcompiler_builtins-b967dc43a87dce64.rlib" "-Tlink.x" "-Bdynamic"
  = note: /home/clement/.xargo/lib/rustlib/thumbv7em-none-eabihf/lib/libcore-12794acabb4ff5e4.rlib(core-12794acabb4ff5e4.core13-fbd2c624a359fb315320b7ce9e430956.rs.rcgu.o): In function `core::panicking::panic_fmt':
          /home/clement/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/src/libcore/panicking.rs:71: undefined reference to `rust_begin_unwind'
          

error: aborting due to previous error

error: Could not compile `i2c_magnetic`.

```

japaric, 2017-12-29:
That looks like ThinLTO breakage. I thought that was already fixed; perhaps they changed something else?

 Does the program link when compiling in release mode?

Could you try disabling ThinLTO? IIRC, you'll have to add `-Z thinlto=false` (note: two arguments) to .cargo/config, under the `target.thumbv7em-none-eabihf.rustflags` property.

clebi, 2017-12-29:
It is compiling and linking in release mode.

I tried to compile in debug mode with the -Z thinlto=n, it did not worked. I have the same error.

japaric, 2017-12-30:
OK, this is actually caused by incremental compilation being enabled by default on unoptimized builds. (I was looking at the wrong place: I bisected and was looking at rustc changes but the incremental stuff was done on the Cargo side -- I did notice a bunch of incremental artifacts appear on the BAD side of the bisect).

@clebi You can disable incremental compilation by adding these lines to your Cargo.toml:

``` toml
[profile.dev]
incremental = false
```

I've known since long that incremental compilation breaks embedded programs in horrible ways (i.e. at link time) so the recommendation has been to not use when it was opt-in; I had no idea it was already considered stable enough to be enable by default. Honestly, the linker error shown above looks *much less worse* than I recall seeing before but unfortunately I have no time to investigate further right now.

cc @alexcrichton (just to make you aware)

clebi, 2017-12-30:
I have disabled the incremental compilation like you said and there is still the same error.

japaric, 2017-12-30:
@clebi Try `cargo clean`? I thought that maybe disabling incremental compilation wasn't making Cargo redo all the build steps but that doesn't seem to be the case, or at least I can't repro with the latest nightly. Could you also double check that incremental compilation *is* disabled? You should **not** have artifacts like these:

``` console
$ tree target
└── thumbv7m-none-eabi
    └── debug
        ├── incremental
        │   └── blue_pill-2xtd0v90b49rb
        │       ├── s-eww3bret4k-1tl2y90-h8om2iji9y08
        │       │   ├── cgu-blue_pill-afio.bc-compressed
        │       │   ├── cgu-blue_pill-afio.o
```

`target/thumbv7m-none-eabi/debug/incremental` should be empty.

clebi, 2017-12-30:
I did the cargo clean.

When I execute `xargo build` the output contains `warning: unused manifest key: profile.dev.incremental` and I still have the same error.

I checked the tree and the incremental folder is empty.

The program I am building is [there](https://github.com/clebi/stm32_magnetic) if it can be of any help. It does not contain all the steps you told me to do.

japaric, 2017-12-30:
> warning: unused manifest key: profile.dev.incremental

Oh, that sounds like you have nightly-2017-12-24 or older. Incremental has not been enabled by default in those nightlies.

I can repro your problem using your repository; it seems to be caused by the now default use of multiple codegen units. I though we fixed something related to that some time ago but maybe it was for the release profile (thinLTO). In any case you can workaroud the issue by using a single codegen unit: add this to Cargo.toml:

``` toml
[profile.dev]
codegen-units = 1
```

clebi, 2017-12-30:
It works with a single codegen unit.

I saw the bug first on rustc 1.24.0-nightly (77e189cd7 2017-12-28) and today on rustc 1.24.0-nightly (2dad872a2 2017-12-29).

Thank you for the help fixing my build.

japaric, 2018-01-17:
As a stopgap solution cortex-m-quickstart v0.2.2 has incremental compilation and parallel codegen disabled in the dev profile. The actual root of the issue is rustc bug (cf. rust-lang/rust#47074) so I'm closing this since it's being tracked upstream (and it should get fixed Soon (TM)).

closed 2018-01-17 by japaric

Samonitari, 2018-01-25:
Thanks, 
I also run into this.

I was porting examples of latest cortex-m-rtfm to a different device crate, so I did not have a thorough look on quickstart crate this time.

Maybe a silly question, if you have time to answer: I know that rtfm examples successfully build in debug.
Why this issue doesn't show up there, although Cargo.toml has no codegen-units setting there?

japaric, 2018-01-25:
> Why this issue doesn't show up there, although Cargo.toml has no codegen-units setting there?

The issue appears to be totally random. It shows up in some projects but not in others. I think it depends on how the compiler ends up partitioning crates into smaller (LLVM) "modules" that then get worked on it parallel (when parallel codegen is enabled).