rust-embedded/cortex-m-rt #52: default_handler no longer includes _e

nallar, 2018-01-07:
In cortex-m, default_handler had a local variable for the exception for convenience when debugging:

https://github.com/japaric/cortex-m/commit/83945819af5a724ba9175fdc5e22b7c854a0fc14#diff-86a9d5942521ed33df4dccb8dbf23016R164

It's missing now that default_handler is in cortex-m-rt:

https://github.com/japaric/cortex-m-rt/blob/8842d0efd7863c2dc23edd75548a658af59afe6b/src/lib.rs#L458

Is this intentional?

japaric, 2018-01-07:
> Is this intentional?

Yes, I don't recall the exact reason but the load operation was not elidable in release mode (because it's a volatile load) yet the variable was not inspectable (in release mode) because of inlining / optimization. This might have been related to the bloating we used to have due to repeated exception / interrupt handlers. (cf. japaric/cortex-m#19)

I think we could provide GDB (Python script) helpers for this debugging stuff since the involved registers are available on all devices and always have the same addresses / semantics. I'm not familiar with GDB scripting though.

If you do want to perform some logic based on the current exception / interrupt in the default handler then you can always override the default handler using the `default_handler!` macro.

japaric, 2018-05-13:
`DefaultHandler` now includes an `irqn` argument which is the IRQ number of the exception being serviced (negative = core exception, zero or positive = device specific interrupt).

closed 2018-05-13 by japaric