rust-embedded/cortex-m-rt #54: Unable to upgrade to cortex-m 0.4.3

clebi, 2018-01-29:
I've tried to update the cortex-m crate to version 0.4.3. 
There is a problem to get peripherals to enable the FPU. I didn't found a solution to get peripherals without taking the sole ownership of it.

The current code is : 
```
let scb = &*cortex_m::peripheral::SCB.get();
scb.enable_fpu();
```

We can't do that because peripherals will not be accessible from other scopes : 

```
let mut scb: cortex_m::peripheral::SCB = cortex_m::Peripherals::steal().SCB;
scb.enable_fpu();
```

Maybe I am missing somethind or we can't upgrade to this version of cortex-m crate.

japaric, 2018-02-07:
> There is a problem to get peripherals to enable the FPU. 

(FWIW, cortex-m-rt enables the FPU before `main` on the thumbv7em-none-eabihf target)

`let p = cortex_m::Peripherals::take().unwrap()` will move all the core peripherals into the current execution context. After that you can pass `p.SCB` around to wherever you need. If you are using rtfm, SCB is under `p.core` where `p: init::Peripherals`.

> We can't do that because peripherals will not be accessible from other scopes :

You can move SCB from `init` to any task using a late resource. If several tasks need access to it then the SCB *resource* can be shared by several tasks.

If any of that doesn't work for you could you give me more details about how your application is structured?

clebi, 2018-02-09:
I've tried to update the cortex-m crate in cortex-m-rt. When switching to cortex-m 0.4.3, I had a problem on the fpu enable part before main. I've looked for a solution and did not find anything that could works without taking preripherals and making them not accessible in main.

GordonHJSmith, 2018-04-01:
I think I have the same issue on cortex-m 0.4.3, unable to run through [ scb.enable_fpu();](https://github.com/japaric/cortex-m-rt/blob/eb9970f9fcd8c42efc4ea2c76cfd908fb6c65366/src/lib.rs#L556) line on a thumbv7em-none-eabihf target (STM32L432). 
Gdb does not give back the hand once I step through this line and I have to kill it.
Everything works fine if I disable the fpu (switch to a thumbv7em-none-eabi target).

jonas-schievink, 2019-12-17:
Is this still an issue? The last comment was about a cortex-m version that still required nightly. A lot has changed since then. Are there any MCUs that require specific FPU enabling actions?

jonas-schievink, 2020-08-30:
Closing this as fixed. c-m-rt now uses assembly to enable the FPU. If there are still issues with the current version, please file a new issue.

closed 2020-08-30 by jonas-schievink