rust-embedded/cortex-m-rt #60: Depend on a no_std compatible version of chrono

robomancer-or, 2018-03-20:
I'm running into a dependency conflict where I can't depend on `uom` because it depends on `num-traits`, which has been made no_std compatible since the most recent version of `chrono`.  The problem is, the released version of `chrono` depends on a conflicting version of `num-traits` that requires `std`.  Why cargo can't sort out conflicts of this sort is beyond me, but since it can't, would `cortex-m-rt` switching to the github version of `chrono` be permissible (at least until `chrono` gets their act together and cuts a no_std compatible release)?

hannobraun, 2018-03-27:
> would `cortex-m-rt` switching to the github version of `chrono` be permissible (at least until `chrono` gets their act together and cuts a no_std compatible release)?

I'd like to note that crates.io doesn't accept crates that depend on git repositories, so switching to chrono's GitHub version would effectively block new versions of cortex-m-rt from being released.

robomancer-or, 2018-03-27:
...Oh, I definitely didn't realize that, yeah, that's definitely not worth it!

skierpage, 2018-04-01:
https://crates.io/crates/chrono is now at 0.4.1 which is no_std compatible. \o/?

whitequark, 2018-04-06:
Note that because of https://github.com/rust-lang/cargo/issues/4866 it is not enough for chrono to support `no_std` to handle every case. Dropping it is the right approach.

commit d8129387 (2018-04-08): drop support of older nightlies

Minor version has been bumped to v0.4.0 so we can stop supporting them

This also drop the chrono dependency, which some people have been running into problems with.

closes #60
closed 2018-04-09 by japaric