rust-embedded/cortex-m-rt #74: LLVM error on High Sierra

chocol4te, 2018-05-16:
Building `cortex-m-rt` on Mac OS High Sierra (tested on two machines) results in the following error:

```
LLVM ERROR: Global variable '__RESET_VECTOR' has an invalid section specifier '.vector_table.reset_vector': mach-o section specifier requires a segment whose length is between 1 and 16 characters.
```

It build correctly on my Ubuntu 18.04 machine.

japaric, 2018-05-16:
> mach-o section specifier requires a segment whose length is between 1 and 16 characters

Are you compiling the `cortex-m-rt` crate for the host (for mac)?

If that's the case we can drop the link_section when compiling for mac; trying to link a cortex-m program compiled for the host (mac / Linux) is going to fail anyway.

chocol4te, 2018-05-16:
I'm so sorry, I'm the biggest idiot. I forgot to set the default target in .cargo/config.

jonas-schievink, 2020-09-04:
The cortex-m crate goes out of its way to build at least on Linux, would be nice if all the core crates consistently built on Linux, macOS, and Windows

japaric, 2021-01-15:
we ran into a similar issue in the `defmt` project.
we found out that on macOS the longest linker section name can be `.section,0123456789ABCDEF` (note: **comma** after `.section` not period and a max of 16 characters after the comma)
this issue can probably be fixed by adding a `cfg_attr(is_macos, link_section = "..")` to the relevant static variables

therealprof, 2021-01-15:
Indeed, I've encountered it frequently.

closed 2021-01-26 by bors[bot]

berkus, 2021-01-27:
What does this fix do? It replaces section name with exactly the same but taking half a kilobyte more space in source code; am I missing something there?

therealprof, 2021-01-27:
@berkus Not sure what you mean. What are you seeing and on which system? This is changing the section names for macOS only because it has a weird length restriction. It should not affect anything except to prevent linker errors on macOS.

berkus, 2021-01-27:
@therealprof ok,  lets go with the way of example, shall we.

```
#[cfg_attr(target_os = "macos", link_section = ".vector_table,interrupts")]
#[cfg_attr(not(target_os = "macos"), link_section = ".vector_table.interrupts")]
```

On  macOS the  section  name would be `.vector_table.interrupts`,  however on any other system it would be `.vector_table.interrupts` which is EXACTLY THE SAME.

So now you've reduced readability, understandability and maintainability of the  code to provide exactly the same section names for every system out there.

What am I missing? I could have missed something, to be precise am looking  at [this](https://github.com/rust-embedded/cortex-m-rt/commit/4aee0941a0448ced77083538b24a6afd16f9f7c7) commit.

Here it  is, condensed:
```
#[cfg_attr(target_os = "macos", link_section = ".HardFault,user")]
#[cfg_attr(not(target_os = "macos"), link_section = ".HardFault.user")] // NB: same name

#[cfg_attr(target_os = "macos", link_section = ".vector_table,reset_vector")]
#[cfg_attr(not(target_os = "macos"), link_section = ".vector_table.reset_vector")] // NB: same name

#[cfg_attr(target_os = "macos", link_section = ".HardFault,default")]
#[cfg_attr(not(target_os = "macos"), link_section = ".HardFault.default")] // NB: same name

#[cfg_attr(target_os = "macos", link_section = ".vector_table,exceptions")]
#[cfg_attr(not(target_os = "macos"), link_section = ".vector_table.exceptions")] // NB: same name

#[cfg_attr(target_os = "macos", link_section = ".vector_table,interrupts")]
#[cfg_attr(not(target_os = "macos"), link_section = ".vector_table.interrupts")] // NB: same name
```

berkus, 2021-01-27:
Ahha, now I see the  comma vs period, my bad!

berkus, 2021-01-27:
@therealprof btw, could it be using the comma on  all systems, not only macos?

therealprof, 2021-01-27:
Not sure sorry, I'm actually curious why this is supposed to work on macOS (NB: I haven't actually tested it myself). Maybe some internal linker magic to substitute the "." back?

berkus, 2021-01-27:
@therealprof on mac this is  two names, ".vector_table" for the segment and "reset_vector" for the section name.

I will test on linux if a section name with comma works, the ELF file format spec says _why not_ so  the only limiting factor would be linker's parser.

adamgreig, 2021-01-27:
We could probably just remove the `link_section` attribute _entirely_ on MacOS, right? It's not going to actually do anything as I understand it, since we can't build final libraries or executables targeting MacOS with cortex-m-rt and it doesn't use our linker script (where the names matter) on MacOS either; this change just lets people on MacOS run cargo check/test without a target specified. I wouldn't want to swap the ARM builds or linker script to using commas for section names just so that Macs can check the code when compiling natively, though.

commit 4aee0941 (2021-01-26): Merge #306

306: fix #74 r=jonas-schievink a=spookyvision

special-cased linker section names to satisfy macos linker constraints
commit 4bbee87b (2021-01-26): fix #74