rust-embedded/cortex-m-rt #75: Confused C Compiler Call

bergus, 2018-05-17:
According to the changelog and docs, since v5

> This crate requires `arm-none-eabi-gcc` to be installed and available in `$PATH`.

I thought that would be no problem, as my setup fulfills this requirement. However, when I try to build, this happens:
```
$ where arm-none-eabi-gcc
C:\Program Files (x86)\GNU Tools ARM Embedded\bin\arm-none-eabi-gcc.exe
$ cargo build
   Compiling demo v0.1.0 (file:///C:/Users/…/demo)
   Compiling cortex-m-rt v0.5.1
error: failed to run custom build command for `cortex-m-rt v0.5.1`
process didn't exit successfully: `C:\Users\…\demo\target\debug\build\cortex-m-rt-4eb45f2a4d8a1423\build-script-build` (exit code: 101)
--- stdout
cargo:rustc-cfg=has_fpu
TARGET = Some("thumbv7em-none-eabihf")
OPT_LEVEL = Some("0")
TARGET = Some("thumbv7em-none-eabihf")
HOST = Some("x86_64-pc-windows-msvc")
TARGET = Some("thumbv7em-none-eabihf")
TARGET = Some("thumbv7em-none-eabihf")
HOST = Some("x86_64-pc-windows-msvc")
CC_thumbv7em-none-eabihf = None
CC_thumbv7em_none_eabihf = None
TARGET_CC = None
CC = Some("x86_64-w64-mingw32-gcc.exe")
TARGET = Some("thumbv7em-none-eabihf")
HOST = Some("x86_64-pc-windows-msvc")
CFLAGS_thumbv7em-none-eabihf = None
CFLAGS_thumbv7em_none_eabihf = None
TARGET_CFLAGS = None
CFLAGS = None
DEBUG = Some("true")
running: "x86_64-w64-mingw32-gcc.exe" "-O0" "-ffunction-sections" "-fdata-sections" "-fPIC" "-g" "-mthumb" "-mfloat-abi=hard" "-march=armv7e-m" "-mfpu=fpv4-sp-d16" "-Wall" "-Wextra" "-o" "C:\\Users\\…\\demo\\target\\thumbv7em-none-eabihf\\debug\\build\\cortex-m-rt-57618b7d7114300a\\out\\asm.o" "-c" "asm.s"
cargo:warning=x86_64-w64-mingw32-gcc: error: unrecognized command line option »-mthumb«
cargo:warning=x86_64-w64-mingw32-gcc: error: unrecognized command line option »-mfloat-abi=hard«
cargo:warning=x86_64-w64-mingw32-gcc: error: unrecognized command line option »-mfpu=fpv4-sp-d16«
exit code: 1

--- stderr
thread 'main' panicked at '

Internal error occurred: Command "x86_64-w64-mingw32-gcc.exe" "-O0" "-ffunction-sections" "-fdata-sections" "-fPIC" "-g" "-mthumb" "-mfloat-abi=hard" "-march=armv7e-m" "-mfpu=fpv4-sp-d16" "-Wall" "-Wextra" "-o" "C:\\Users\\…\\demo\\target\\thumbv7em-none-eabihf\\debug\\build\\cortex-m-rt-57618b7d7114300a\\out\\asm.o" "-c" "asm.s" with args "x86_64-w64-mingw32-gcc.exe" did not execute successfully (status code exit code: 1).

', C:\Users\…\.cargo\registry\src\github.com-1ecc6299db9ec823\cc-1.0.15\src\lib.rs:2158:5
note: Run with `RUST_BACKTRACE=1` for a backtrace.

warning: build failed, waiting for other jobs to finish...
error: build failed
```
It seems that you're using the [`cc` crate](https://crates.io/crates/cc) to invoke the compilation of some assembly, but for some reason it selects `x86_64-w64-mingw32-gcc` (which doesn't know the arm options) instead of `arm-none-eabi-gcc`. I have studied the options of `cc`, but I'm not sure whether I need to setup my environment differently or whether there is a bug in the build script.

japaric, 2018-05-17:
> CC = Some("x86_64-w64-mingw32-gcc.exe")

Apparently you have the CC variable set in your environment. For some reason the `cc` crate will use the value of this variable when cross compiling. Personally, I think it shouldn't do that. We should file an issue in the cc repository about using CC when cross compiling.

bergus, 2018-05-17:
Ah, thanks! I had assumed that those stdout statements all were generated from the project settings, not being influenced from my environment.  I think my Ocaml setup required that variable, I'll delete it as I no longer use Ocaml.
Without the `CC` environment variable, it works perfectly!

closed 2018-05-17 by bergus