rust-embedded/cortex-m-rt #76: The reset, exception, and interrupt vectors are missing.

jcreedon, 2018-05-18:
I am getting the following when trying to build my application.

          cortex-m-rt: The reset vector is missing. This is a bug in cortex-m-rt. Please file a bug
          report at: https://github.com/japaric/cortex-m-rt/issues
          /usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld:
          cortex-m-rt: The exception vectors are missing. This is a bug in cortex-m-rt. Please file
          a bug report at: https://github.com/japaric/cortex-m-rt/issues
          /usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld:
          cortex-m-rt: The interrupt vectors are missing. Possible solutions, from most likely to
          less likely:
          - Link to a device crate
          - Disable the 'device' feature of cortex-m-rt to build a generic application (a dependency
            may be enabling it)
          - Supply the interrupt handlers yourself. Check the documentation for details.

After running `cargo rustc --target thumbv7em-none-eabihf -- -C link-arg=-nostartfiles -C link-arg=-Tlink.x` per the documentation [here](https://docs.rs/cortex-m-rt/0.5.1/cortex_m_rt/#an-example).

Per the provided instruction I am filing a bug here.

I tried disabling the `device` feature (by way of disabling the `rt` feature on the device crate) and that did not work. My code is here https://github.com/jcreedon/bitsy-blinky and my device crate is here https://github.com/jcreedon/stm32f41x (generated on 0.13.0)

japaric, 2018-05-18:
TL;DR Use `cargo build` not `cargo rustc ..`.

> After running cargo rustc --target thumbv7em-none-eabihf -- -C link-arg=-nostartfiles -C link-arg=-Tlink.x per the documentation here.

Your project has a .cargo/config file which the cortex-m-rt documentation doesn't mention. Were you following the [cortex-m-quickstart] instructions?

[cortex-m-quickstart]: https://docs.rs/cortex-m-quickstart/0.3.1/cortex_m_quickstart/

If you have a .cargo/config that adds `-nostartfiles -Wl,-Tlink.x` to the linker arguments then you don't to use `cargo rustc ..`; `cargo build` is enough. If you use `cargo rustc ..` you'll end up passing the linker script (`link.x`) *twice* to the linker; which is what causes the linker errors you see.

jcreedon, 2018-05-18:
`cargo build` completes successfully without errors, but the image generated will not load. It fails on `.vector_table`. I was working my way backwards through the stack debugging things and was trying to manually build to see if that produced anything different. I did not realize that calling `cargo rustc` was still using `.cargo/config`.

I'll keep digging around.

closed 2018-05-18 by jcreedon