rust-embedded/cortex-m-rt #80: cortex-m-rt doesn't build on Windows anymore

jcreedon, 2018-07-03:
On Windows, with a fresh clean install of `arm-none-eabi-gcc`, building `cortex-m-rt` eventually fails with:

```
Internal error occurred: Failed to find tool. Is `ar` installed? (see https://github.com/alexcrichton/cc-rs#compile-time-requirements for help)
```

As a workaround, setting the `TARGET_AR` environment variable to `arm-none-eabi-ar` fixes it. This was not necessary before, and is not documented anywhere if it is a new requirement.

tib888, 2018-07-15:
I had the same issue on windows.
Thank you for this fix!

japaric, 2018-08-02:
IMO, `cc` should be using a prefixed archiver if already using a prefixed compiler / assembler. I have filed alexcrichton/cc-rs#330 to request changing the behavior.

jonathanpallant, 2020-05-19:
I fell over this today :/

closed 2020-08-26 by bors[bot]