rust-embedded/cortex-m-rt #89: Document the minimum GCC / LD supported version

japaric, 2018-08-18:
As we found out in #84 our linker script is not supported by GCC 4.8 / binutils 2.24. We should document the minimum version that we support.

I know that binutils 2.31 works.

mullr, 2018-08-22:
2.27, the one currently in ubuntu 18.04, appears not to work. (gives the same error as #92)

mvirkkunen, 2018-08-26:
Also broken on GCC 6.3.1 which is the current version of gcc-arm-none-eabi in the official Ubuntu repositories. Same error as #92. I switched to LLD for the time being.

therealprof, 2018-08-27:
@adamgreig Funny you should mention this. I also had various troubles in the past specifically with 2016q3... I would definitely second that we recommend using the latest ARM release, that'll reduce the amount of uncertainty considerably.

japaric, 2018-08-30:
No longer needed because this crate no longer depends on arm-none-eabi-gcc. :tada:

closed 2018-08-30 by japaric