rust-embedded/cortex-m-rt #92: Linker errors with 0.5.2 and arm-none-eabi-gcc

phil-opp, 2018-08-20:
When I try to link [embed-rs/stm32f7-dicovery-rewrite](https://github.com/embed-rs/stm32f7-discovery-rewrite) with arm-none-eabi-gcc 4.9.3, I get the following error with cortex-m-rt 0.5.2:

```
error: linking with `arm-none-eabi-gcc` failed: exit code: 1
  |
  = note: "arm-none-eabi-gcc" "-L" […]
  = note: /usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: 
          BUG(cortex-m-rt): the reset vector is missing
          /usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: 
          BUG(cortex-m-rt): the exception vectors are missing
          collect2: error: ld returned 1 exit status
```

With cortex-m-rt 0.5.1, it builds without problems.

adamgreig, 2018-08-20:
See #89 - the new linker script (changed to support LLD) doesn't support old GCC 4.8, and perhaps also therefore 4.9 which you're using. Are you able to update to a newer GCC? 4.9.3 is from 2015 and the latest is 8.2; you can grab pre-built images directly from ARM so upgrading is usually fairly painless.

Alternatively you might be able to swap to using LLD and thereby not require an ARM GCC toolchain at all...

phil-opp, 2018-08-20:
Thanks for the quick answer! I'll try if I can get LLD working.

Perhaps it's a good idea to bump the version to 0.6.0 if it is breaking for some users. It was time consuming for me to find out which crate introduced the breakage after I ran `cargo update`.

phil-opp, 2018-08-20:
Linking with LLD works fine. Thanks!

closed 2018-08-20 by phil-opp