rust-embedded/cortex-m-rt #105: New cortex-m-rt rand dependency breaks builds

therealprof, 2018-09-06:
I was just trying to port the `microbit` crate to `cortex-m-rt` 0.6.0 but I'm facing this here:
```
   Compiling rand_core v0.2.1
   Compiling cortex-m-rt v0.6.0
error[E0463]: can't find crate for `std`
  |
  = note: the `thumbv6m-none-eabi` target may not be installed

error: aborting due to previous error

For more information about this error, try `rustc --explain E0463`.
error: Could not compile `rand_core`.
warning: build failed, waiting for other jobs to finish...
error: build failed
```

@japaric Seems something is messing (or not) with the required flags for `no_std`.

japaric, 2018-09-06:
*sigh* this is probably that Cargo limitation where it keeps a single dependency graph for both build-dependencies (compiled for the host) and other dependencies (compiled for the target). Making `rand` not depend on `std` *may* fix this.

japaric, 2018-09-06:
> Making rand not depend on std may fix this.

I meant our `rand` build dependency not depend on std may fix this.

therealprof, 2018-09-06:
That works. NB: It's fairly annoying to test that... requires all local checkouts and changing everything to paths because (of course) `cortex-m-rt` is first imported by the peripheral access API crates...

Oh, and I run into another funkiness which prevents me from compiling `cortex-m-rt` examples from the checkout:
```
   Compiling volatile-register v0.2.0
error[E0658]: use of unstable library feature 'rustc_private': this crate is being loaded from the sysroot, an unstable location; did you mean to load this crate from crates.io via `Cargo.toml` instead? (see issue #27812)
 --> macros/src/lib.rs:4:1
  |
4 | extern crate rand;
  | ^^^^^^^^^^^^^^^^^^
  |
  = help: add #![feature(rustc_private)] to the crate attributes to enable

error[E0658]: use of unstable library feature 'rustc_private': this crate is being loaded from the sysroot, an unstable location; did you mean to load this crate from crates.io via `Cargo.toml` instead? (see issue #27812)
...
```

adamgreig, 2018-09-06:
I _don't_ get this error when using the new 0.6.0 on a thumbv7em-none-eabi target. Everything builds fine.

japaric, 2018-09-06:
@adamgreig this only happens if your crate depend on the `rand` crate.

adamgreig, 2018-09-06:
Right, I'd just realised but you beat me to pointing it out. Makes sense.

therealprof, 2018-09-06:
Ah, and I get your point now. Will do. ;)

therealprof, 2018-09-07:
Can we move this forward?

little-arhat, 2018-09-08:
What is the plan for this? Will merging https://github.com/therealprof/cortex-m-rt/commit/bddf23b3fa73d2d3abfc7491e352ac106a6111b3 suffice or is there more to it?

commit bddf23b3 (2018-09-08): Disable default-features on rand dependency to avoid std version

Due to the single dependency tree, the attempted use of a std version
flips all depending crates to the std version as well which will not
compile on no_std systems.

Fixes #105
commit 082f9074 (2018-09-08): Merge #107

107: Disable default-features on rand dependency to avoid std version r=japaric a=therealprof

Due to the single dependency tree, the attempted use of a std version
flips all depending crates to the std version as well which will not
compile on no_std systems.

Fixes #105
PR #107 (2018-09-08): Disable default-features on rand dependency to avoid std version
Due to the single dependency tree, the attempted use of a std version
flips all depending crates to the std version as well which will not
compile on no_std systems.

Fixes #105
closed 2018-09-08 by bors[bot]