rust-embedded/cortex-m-rt #109: How to define interrupt handlers?

therealprof, 2018-09-07:
Now that we have the fantastic new `#[entry]` and `#[exception]` macros, how do we deal with interrupt handlers?

japaric, 2018-09-07:
Ideas:

1. When the "device" feature is not enabled and cortex-m-rt registers 32 (v6-M) or 240 (v7-M) interrupts in the vector table we could have an `#[interrupt]` attribute that uses *position* rather than name:

``` rust
// strawman syntax
#[interrupt(0)] // first interrupt handler
fn foo() {}

#[interrupt(1)] // second interrupt handler
fn bar() {}

#[interrupt(240)] // this is an error
fn baz() {}
```

2. When the "device" feature is enabled and the device crate provides N named interrupt we change the syntax of the `#[interrupt]` attribute to:

``` rust
#[interrupt]
fn USART1() {}
```

To check that USART1 is a valid interrupt name we could (ab)use the fact that attributes are *not* in the type namespace to have e.g. `use stm32f103xx::interrupt` import *both* the attribute and an enumeration of valid interrupt names. Then the expansion of `#[interrupt]` would use the imported interrupt to check if it's valid or not. With code:

``` rust
// crate: stm32f103xx

extern crate cortex_m_rt;

pub use cortex_m_rt::_interrupt as interrupt; // this is the attribute

// this is the enumeration of valid interrupt names
enum interrupt { USART1, USART2, .. }
```

Then this

``` rust
// user code

use stm32f103xx::interrupt; // imports both the attribute and the enumeration

#[interrupt]
fn USART1() {
    // code goes here
}
```

would expand into

``` rust
// expanded user code

use stm32f103xx::interrupt;

#[export_name = "USART1"]
pub extern "C" fn some_random_hash() {
    interrupt::USART1;  // checks that USART1 exists in the enumeration

    // code goes here
}
```

The downside of this hack is that this wouldn't work:

``` rust
#[stm32f103xx::interrupt]
fn USART1() {
    // code goes here
}
```

because the interrupt enum has not been imported. Though I don't think it's idiomatic to use attributes without importing them first.

I haven't actually tested that this last part about the check works but I think it should work.

Also, we don't necessarily have to do (1) *and* (2). We could do only (2).

therealprof, 2018-09-08:
Very nice, I like it, especially idea number 2. Can't we simply use the `Nr` trait from `bare_metal` to allow for custom interrupt mapping (i.e. the implementation of idea number 1)?

PS: The new macros are absolutely fabulous: both extremely intuitive and simple to use and far less boilerplatey. If we can extend that to interrupts as well we have a fantastic new attraction at hand.

korken89, 2018-09-08:
This look very intuitive, I like it!

adamgreig, 2018-09-09:
I like it, and it removes the `interrupt!` macro from the device crates and makes it just a thing in c-m-rt, which works around the worry of having to have a macros crate for each device crate.

> Also, we don't necessarily have to do (1) and (2). We could do only (2).

Could we do both? If you specify an argument it's the interrupt number and is used, otherwise you have to have the relevant interrupt enum imported and the name must match up. It means the attribute in c-m-rt doesn't require use of a device crate if you're not using one. Support for the enum variant could still require the `device` feature. Best of both worlds?

adamgreig, 2018-10-09:
Closed by #122.

closed 2018-10-09 by adamgreig