rust-embedded/cortex-m-rt #110: RAM initialization routines are huge

japaric, 2018-09-07:
when compiling with optimizations because LLVM loop unrolls zero_bss and init_data. This causes binaries to be about 500 bytes bigger in size compared to non-loop unrolled version. Here are the disassemblies:

NOTE: I have inserted BKPTs before and after zero_bss and init_data

- `opt-level = "z"`

``` console
$ size -A app
.text                 78       0x400
```

``` armasm
00000400 <Reset>:
 400:   f000 f820       bl      444 <DefaultPreInit>
 404:   4809            ldr     r0, [pc, #36]   ; (42c <_stext+0x2c>)
 406:   2200            movs    r2, #0
 408:   4909            ldr     r1, [pc, #36]   ; (430 <_stext+0x30>)
 40a:   be00            bkpt    0x0000
 40c:   e000            b.n     410 <Reset+0x10>
 40e:   c004            stmia   r0!, {r2}
 410:   4288            cmp     r0, r1
 412:   d3fc            bcc.n   40e <Reset+0xe>
 414:   4807            ldr     r0, [pc, #28]   ; (434 <_stext+0x34>)
 416:   be00            bkpt    0x0000
 418:   4907            ldr     r1, [pc, #28]   ; (438 <_stext+0x38>)
 41a:   4a08            ldr     r2, [pc, #32]   ; (43c <_stext+0x3c>)
 41c:   e001            b.n     422 <Reset+0x22>
 41e:   c908            ldmia   r1!, {r3}
 420:   c008            stmia   r0!, {r3}
 422:   4290            cmp     r0, r2
 424:   d3fb            bcc.n   41e <Reset+0x1e>
 426:   be00            bkpt    0x0000
 428:   e7fe            b.n     428 <Reset+0x28>
 42a:   bf00            nop
 42c:   20000000        .word   0x20000000
 430:   20000000        .word   0x20000000
 434:   20000000        .word   0x20000000
 438:   00000450        .word   0x00000450
 43c:   20000000        .word   0x20000000
```

- `opt-level = "3"` (default)

``` console
$ size -A app
.text                594       0x400
```

``` armasm
00000400 <Reset>:
 400:	f000 f922 	bl	648 <DefaultPreInit>
 404:	f240 0000 	movw	r0, #0
 408:	f240 0200 	movw	r2, #0
 40c:	f2c2 0000 	movt	r0, #8192	; 0x2000
 410:	f2c2 0200 	movt	r2, #8192	; 0x2000
 414:	4282      	cmp	r2, r0
 416:	be00      	bkpt	0x0000
 418:	d25a      	bcs.n	4d0 <Reset+0xd0>
 41a:	1d11      	adds	r1, r2, #4
 41c:	ea6f 0c02 	mvn.w	ip, r2
 420:	4281      	cmp	r1, r0
 422:	f04f 0401 	mov.w	r4, #1
 426:	460b      	mov	r3, r1
 428:	bf38      	it	cc
 42a:	4603      	movcc	r3, r0
 42c:	4463      	add	r3, ip
 42e:	eb04 0393 	add.w	r3, r4, r3, lsr #2
 432:	079b      	lsls	r3, r3, #30
 434:	4613      	mov	r3, r2
 436:	d024      	beq.n	482 <Reset+0x82>
[... 182 lines cut ...]
 61a:	d2c8      	bcs.n	5ae <Reset+0x1ae>
 61c:	e7ed      	b.n	5fa <Reset+0x1fa>
 61e:	f104 0108 	add.w	r1, r4, #8
 622:	f100 0308 	add.w	r3, r0, #8
 626:	e79b      	b.n	560 <Reset+0x160>
 628:	f101 0520 	add.w	r5, r1, #32
 62c:	3130      	adds	r1, #48	; 0x30
 62e:	cd35      	ldmia	r5, {r0, r2, r4, r5}
 630:	6218      	str	r0, [r3, #32]
 632:	625a      	str	r2, [r3, #36]	; 0x24
 634:	f103 0230 	add.w	r2, r3, #48	; 0x30
 638:	629c      	str	r4, [r3, #40]	; 0x28
 63a:	62dd      	str	r5, [r3, #44]	; 0x2c
 63c:	f1bc 0f30 	cmp.w	ip, #48	; 0x30
 640:	d2b5      	bcs.n	5ae <Reset+0x1ae>
 642:	e7da      	b.n	5fa <Reset+0x1fa>
```

Possible solutions:

- Recommend using "z". I don't like this solution because optimizations for size penalize inlining and can cause other code to not be properly optimized.

- Implement zero_bss and init_data using assembly making them tight loops like the ones you see in the output of `opt-level = "z"`.

Thoughts @rust-embedded/cortex-m ?

japaric, 2018-09-07:
@therealprof I'm not sure I follow your comment. This issue is orthogonal to ramfuncs; all programs are affected by this bloat because .bss and .data are always initialized, even if they are empty.

therealprof, 2018-09-07:
@japaric My bad, I thought you were referring to a new problem caused by recent changes rather than a pre-existing issue.

I think the main issue here is that there's no distinction between embedded devices and regular machines, there has never been a good "one fits all" approach to defaults and here it is more true than ever. I don't quite agree with your assessment here that recommending a different optimisation level is a bad idea since pretty much all major embedded projects outside of Rust optimise embedded code for size and not for speed (and even chose size over debuggability). When you have a major size constraint it is in my opinion crucial to worry about size first and consider speed later, potentially fine tuning the code generation with explicit inlining (or general optimisation for speed).

Probably we should think about changing default optimisation targets to match reality to yield size optimal results out-of-the-box.

This particular case is still very interesting since this code looks slower to me on first glance, as if the compiler tried to be especially clever by mostly unrolling the loop to avoid misalignment penalties and badly predicted branches and whatnot... Which architecture what that compiled for? Also I wonder whether Rust provides ways to annotate code to tell the compiler that this function is only called once and thus should never be subject to ridiculous speed optimisations... I've only found `#[cold]` at a glance, do we have a change to apply that?

japaric, 2018-09-07:
> Probably we should think about changing default optimisation targets to match reality to yield size optimal results out-of-the-box.

As I mentioned above optimizing for size reduces the inlining threshold. This can cause the compiler to not be able to optimize away dead branches which can result in binaries bigger than the ones optimized for speed. I have seen this with LLVM 4: opt-level = 3 consistently produced smaller binaries than "s" and "z". Then things changed with the move to LLVM 5 because of the aggressive loop unrolling seen in this issue -- the bloat makes binaries compiled with "3" larger than "s" and "z" binaries.

> Which architecture what that compiled for?

Cortex-M3 / `thumbv7m-none-eabi`

> Also I wonder whether Rust provides ways to annotate code to tell the compiler that this function is only called once and thus should never be subject to ridiculous speed optimisations

That would be the `optimize` attribute proposed in rust-lang/rfcs#2412 but it doesn't exist right now.

>  I've only found #[cold] at a glance, do we have a change to apply that?

I tried that and it had no effect on the binary size.

adamgreig, 2018-09-07:
I think we should ensure the initialisation routines are small and effective independent of the opt-level anyway; plenty of embedded targets have megabytes of flash and care more about speed; others have a couple kilobytes and don't care at all about speed. Recommending a specific opt-level in this very foundational crate seems ill advised to me. We should have a good r0 independently of the opt-level.

therealprof, 2018-09-07:
> I don't think DMA would be much quicker than CPU, unless you set the DMA off doing one thing while the CPU did another.

It's much quicker for bulk operations. The DMA engine can execute copies at bus speed while a copy loop is at least 4 clock cycles but that point is moot. My point was: There might be faster ways to copy than a decrement counter by 4, load 32 bit, store 32 bit, branch and by going assembler you're locked into the implementation (i.e. the supposedly most size efficient one), even if you need something else.

japaric, 2018-09-07:
DMA? This discussion seems to have gone rather off topic. In any case, ...

>  by going assembler you're locked into the implementation

How so? Assuming that we want to add a new feature that lets the user override the RAM initialization routine how does the default behavior being X or Y prevent the user from setting the new behavior to be A or B? I don't see any issue there.

korken89, 2018-09-08:
This is quite interesting as I was struggling with the same issue, but my old embedded roots were just "use optimize for size, it is what we always do when there is bloat", which indeed worked as expected.
And now I am using this optimization in all projects.

Though recommending a default is very much up to the project. For example a bootloader would probably use "z", while a number cruncher would use "3".
Though, as it was discussed in #111 , by removing the bloat from init in all cases only the user space code matters, making testing different flags easy for the users (init code has small impact). And hence I would say the the default is no longer that important. Those who really needs some specific optimization knows what they are doing and what they want :)

korken89, 2018-09-08:
> This compiler bug is only deemed an issue exactly because a lot of people do not know what the default options for --release mode are and they need tweaking for sane results

Exactly, hence #111 solves this.

japaric, 2018-09-18:
I took the time to actually measure the loop unrolled version and, to me surprise, it does make a significant difference in execution time. I tested this on both a Cortex-M3 (STM32F103) and a Cortex-M4F (STM32F303) core. The results were consistent: the unrolled version initializes 4 KiB of RAM in about half the time it takes the non-unrolled version. Numbers at the bottom.

[This] is the test code I used. I executed the program with the debugger detached to rule out it messing with the measurement. If you find any flaw in the measurement let me know. I encourage you to do replicate the measurements.

[This]: https://gist.github.com/japaric/f1422b00e0f6229e973787c50a613415

Other observations:

- opt-level = 2 and opt-level = 3 (--release) do loop unrolling. There was no difference in performance or binary size.

- Loop unrolling costs about 464 bytes. It's done twice in this case: once in zero_bss and again in init_data.

- opt-level = 3 *without* loop unrolling (I patched the compiler to get this mode) does significantly better than opt-level = s and opt-level = z. Execution time was reduced by 15-20%.

What to do now? I'm not exactly sure. I still would like to see more knobs to control loop unrolling, though.

<details>

### 4 KiB .bss

- opt-level = "z"

.text 58 bytes
time  5157 ticks

- opt-level = "s"

.text 82 bytes
time  5150 ticks

- opt-level = 2 +unroll-loops

.text 546 bytes
time  2034 ticks

- --release (opt-level = 3) +unroll-loops

.text 546 bytes
time  2034 ticks -- 39.44% of "z"

- --release (opt-level = 3) -unroll-loops (patched)

.text 82 bytes
time  4123 ticks -- 79.94% of "z"

### 4 KiB .data

- opt-level = "z"

.text 58 bytes
time  7205 ticks

- opt-level = "s"

.text 82 bytes
time  7199 ticks

- opt-level = 2 +unroll-loops

time  3481 ticks

- --release (opt-level = 3) +unroll-loops

.text 546 bytes
time  3481 ticks -- 48.31% of "z"

- --release (opt-level = 3) -unroll-loops (patched)

.text 82 bytes
time  6173 ticks -- 85.67% of "z"
</details>

korken89, 2018-09-18:
This is expected as my calculations shows, at about 4x unrolling you will reach this speed. . :)

therealprof, 2018-09-18:
NB: This code is not (just) unrolled, it's more or less vectorized and the branches are there to deal with with odd sizes and addresses, that's why it is faster:

```
 5ae:	e891 0089 	ldmia.w	r1, {r0, r3, r7}
 5b2:	e9d1 6503 	ldrd	r6, r5, [r1, #12]
 5b6:	e9d1 4c05 	ldrd	r4, ip, [r1, #20]
 5ba:	f8d1 801c 	ldr.w	r8, [r1, #28]
 5be:	e882 0089 	stmia.w	r2, {r0, r3, r7}
 5c2:	f101 0720 	add.w	r7, r1, #32
 5c6:	e9c2 6503 	strd	r6, r5, [r2, #12]
 5ca:	e9c2 4c05 	strd	r4, ip, [r2, #20]
 5ce:	f8c2 801c 	str.w	r8, [r2, #28]
```

korken89, 2018-09-22:
Indeed, I was more towards using the memory buss to its maximum capacity.