rust-embedded/cortex-m-rt #115: New cortex-m-rt doesn't work with Cargo "rename-dependency" feature

little-arhat, 2018-09-09:
In `2018` edition instead of `extern crate cortex_m_rt as rt` one would use Cargo `rename-dependency`:

```
[dependencies]
rt = {version = "0.6.2", package='cortex-m-rt'}
```

cortex_m_rt_macros, however, doesn't respect this and still generates code with `extern crate cortex_m_rt`.

japaric, 2018-09-09:
I haven't been following closely but isn't this a known issue? What does e.g. serde-derive do in the case you rename serde to something else?

little-arhat, 2018-09-09:
It seems `serde_derive` also struggles with renames: https://github.com/serde-rs/serde/issues/953. 
I think there should be at least mention about this behaviour in the doc as currenaly all example in `rust-embedded` use `extern crate cortex_m_rt as rt`, and in 2018 this is done via renames.