rust-embedded/cortex-m-rt #117: Serious compile-time regression (1-2 min to 1 hour)

korken89, 2018-09-10:
Hi, with the latest changes here I have seen a compile time regression from 1-2 min to now a little over 1 hour.
I am not sure what caused the regression, but it can be tested there (I have reproduced it on the latest nightly on both a Ubuntu 16.04 and Arch Linux distribution):
https://github.com/korken89/trustflight_firmware - enter the binary folder and run `cargo build --release`.

The time is mostly in @adamgreig's `stm32f3`, which at the update to `0.2.3` only updated the `cortex-m-rt` version (at `0.2.2` this did not happen), leading me to place the issue here.

Does anyone know what caused this to happen and a possible mitigation?
Thanks!

Rustc: `rustc 1.30.0-nightly (2d4e34ca8 2018-09-09)`

adamgreig, 2018-09-10:
I'm seeing the same compile-time explosion even pinning cortex-m-rt to 0.5.4, so I'm not sure it's actually this crate's fault. Next up is bisecting nightlies. For reference rustc is spending all its time doing something like...

```
#0  core::slice::memchr::contains_zero_byte () at libcore/slice/memchr.rs:33
#1  core::slice::memchr::memchr () at libcore/slice/memchr.rs:80
#2  0x00007f88588729a9 in syntax::source_map::SourceMap::span_to_margin () from /home/adam/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/bin/../lib/../lib/libsyntax-7135e345cfdac322.so
#3  0x00007f8858785cb6 in syntax::parse::lexer::tokentrees::<impl syntax::parse::lexer::StringReader<'a>>::parse_token_tree ()
   from /home/adam/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/bin/../lib/../lib/libsyntax-7135e345cfdac322.so
#4  0x00007f8858785aed in syntax::parse::lexer::tokentrees::<impl syntax::parse::lexer::StringReader<'a>>::parse_token_tree ()
   from /home/adam/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/bin/../lib/../lib/libsyntax-7135e345cfdac322.so
#5  0x00007f8858785aed in syntax::parse::lexer::tokentrees::<impl syntax::parse::lexer::StringReader<'a>>::parse_token_tree ()
   from /home/adam/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/bin/../lib/../lib/libsyntax-7135e345cfdac322.so
#6  0x00007f8858785aed in syntax::parse::lexer::tokentrees::<impl syntax::parse::lexer::StringReader<'a>>::parse_token_tree ()
   from /home/adam/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/bin/../lib/../lib/libsyntax-7135e345cfdac322.so
#7  0x00007f88587853cb in syntax::parse::lexer::tokentrees::<impl syntax::parse::lexer::StringReader<'a>>::parse_all_token_trees ()
   from /home/adam/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/bin/../lib/../lib/libsyntax-7135e345cfdac322.so
#8  0x00007f88588779c8 in syntax::parse::source_file_to_stream () from /home/adam/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/bin/../lib/../lib/libsyntax-7135e345cfdac322.so
#9  0x00007f8858877565 in syntax::parse::source_file_to_parser () from /home/adam/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/bin/../lib/../lib/libsyntax-7135e345cfdac322.so
#10 0x00007f88586e19af in syntax::parse::parser::Parser::parse_item_implementation () from /home/adam/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/bin/../lib/../lib/libsyntax-7135e345cfdac322.so
#11 0x00007f88586d40af in syntax::parse::parser::Parser::parse_item_ () from /home/adam/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/bin/../lib/../lib/libsyntax-7135e345cfdac322.so
#12 0x00007f88586e99e9 in syntax::parse::parser::Parser::parse_item () from /home/adam/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/bin/../lib/../lib/libsyntax-7135e345cfdac322.so
#13 0x00007f88586d16d3 in syntax::parse::parser::Parser::parse_mod_items () from /home/adam/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/bin/../lib/../lib/libsyntax-7135e345cfdac322.so
#14 0x00007f88586ea896 in syntax::parse::parser::Parser::parse_crate_mod () from /home/adam/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/bin/../lib/../lib/libsyntax-7135e345cfdac322.so
#15 0x00007f8858876df8 in syntax::parse::parse_crate_from_file () from /home/adam/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/bin/../lib/../lib/libsyntax-7135e345cfdac322.so
<snip>
```

adamgreig, 2018-09-10:
I can build stm32f3 fine with nightly-2018-08-30 (not bisected further) even with latest cortex-m-rt. So this is an issue but it's not cortex-m-rt's issue, I don't think.

Looks like it's https://github.com/rust-lang/rust/issues/54083

closed 2018-09-10 by adamgreig

korken89, 2018-09-11:
Thanks for finding!