rust-embedded/cortex-m-rt #125: Regression on 0.6 on rust messages with attribute macro

TeXitoi, 2018-09-18:
Hi!

When using the new attribute macro (`#[entry] fn main() -> ! { loop {} }`, I have a big regression when compiling and using a debugger: everything (error, warning and debugger) point to the attribute line, making debugging tricky.

A workaround is to directly call another function, but that's quite dirty:
```rust
#[entry]
fn entry() -> ! {
    main()
}
fn main() -> ! {
    loop {}
}
```

adamgreig, 2018-09-18:
This is caused by [#90 (comment)](https://github.com/rust-embedded/cortex-m-rt/pull/90#issuecomment-414086570), in other words a problem in proc-macro2. I think the hope is proc-macro2 will land the required feature soon.

TeXitoi, 2018-09-18:
See https://github.com/alexcrichton/proc-macro2/issues/104 and https://github.com/rust-lang/rust/issues/43081 for reference.

TeXitoi, 2018-09-18:
Maybe updating to syn 0.15 will fix this problem: https://github.com/dtolnay/syn/releases/tag/0.15.0

japaric, 2018-09-19:
@TeXitoi do you have steps to reproduce the problem you are seeing? That way we can evaluate if moving to syn 0.15.0 helps or not.

in particular, I'm seeing that debugging works as intended on 1.30-beta. This is using the dev profile.

[image]

What version of GDB are you using?

Compile errors on statements also have the correct span.

``` rust
// ..

#[entry]
fn main() -> ! {
    let x: () = 42;

    loop {}
}
```

``` console
$ cargo build
error[E0308]: mismatched types
  --> src/main.rs:14:17
   |
14 |     let x: () = 42;
   |                 ^^ expected (), found integral variable
   |
   = note: expected type `()`
              found type `{integer}`
```

Where I see incorrect spans is when e.g. the `#[entry]` function doesn't type check.

``` rust
// ..

#[entry]
fn main() -> ! {}
```

```
$ cargo build
error[E0308]: mismatched types
 --> src/main.rs:9:1
  |
9 | #[entry]
  | ^^^^^^^^ expected !, found ()
  |
  = note: expected type `!`
             found type `()`
```

---

> This is caused by #90 (comment)

Enabling the nightly feature of proc-macro2 should be documented in the crate level documentation.

TeXitoi, 2018-09-20:
I'm on debian stable, gdb-arm-none-eabi 7.12-6+9+b2

But I can't reproduce anymore the error. Maybe a bad nightly.

I close this, and I'll reopen if I manage to reproduce.

johkra, 2018-10-06:
I can reproduce this issue and have put code to reproduce this at https://github.com/johkra/repro.

If I declare the i2c mode inline, cargo check shows the following output:
```
$ cargo check
   Compiling proc-macro2 v0.4.20                                                                         
   Compiling unicode-xid v0.1.0                                                                          
   Compiling rand_core v0.3.0                                                                            
   Compiling cortex-m-rt v0.6.4                                                                          
   Compiling cortex-m-semihosting v0.3.1                                                                 
   Compiling cortex-m v0.5.7                                                                             
    Checking vcell v0.1.0                                                                                
    Checking r0 v0.2.2                                                                                   
    Checking aligned v0.2.0                                                                              
    Checking bare-metal v0.2.3                                                                           
   Compiling stm32f103xx v0.10.0                                                                         
    Checking void v1.0.2                                                                                 
   Compiling stm32f103xx-hal v0.1.0 (https://github.com/ilya-epifanov/stm32f103xx-hal?branch=i2c-support#3a8d64d3)
    Checking nb v0.1.1                                                                                   
    Checking cast v0.2.2                                                                                 
    Checking panic-abort v0.3.1                                                                          
   Compiling rand_core v0.2.2                                                                            
    Checking volatile-register v0.2.0                                                                    
    Checking embedded-hal v0.2.1                                                                         
   Compiling rand v0.5.5                                                                                 
    Checking panic-itm v0.2.0                                                                            
    Checking panic-semihosting v0.3.0                                                                    
   Compiling quote v0.6.8                                                                                
   Compiling syn v0.15.8                                                                                 
   Compiling cortex-m-rt-macros v0.1.2                                                                   
    Checking cortex-m-rt v0.5.7                                                                          
    Checking repro v0.1.0 (/home/johkra/dev/repro)                                                       
warning: unused variable: `i2c`                                                                          
  --> src/main.rs:20:1                                                                                   
   |                                                                                                     
20 | #[entry]                                                                                            
   | ^ help: consider using `_i2c` instead                                                               
   |                                                                                                     
   = note: #[warn(unused_variables)] on by default                                                       
                                                                                                         
warning: function is never used: `get_mode`                                                              
  --> src/main.rs:16:1                                                                                   
   |                                                                                                     
16 | fn get_mode() -> i2c::Mode {                                                                        
   | ^^^^^^^^^^^^^^^^^^^^^^^^^^                                                                          
   |                                                                                                     
   = note: #[warn(dead_code)] on by default                                                              
                                                                                                         
    Finished dev [unoptimized + debuginfo] target(s) in 33.70s    
```

Note how the warning about the unused variable i2c points to the entry macro instead of the actual line.

If I perform a slight change and return the mode declaration from a function instead of inline, the warning points to the correct place.

Diff:
```
$ git diff
diff --git a/src/main.rs b/src/main.rs
index f914728..149f75b 100644
--- a/src/main.rs
+++ b/src/main.rs
@@ -33,8 +33,8 @@ fn main() -> ! {
     let i2c = i2c::BlockingI2c::i2c2(
         dp.I2C2,
         (scl, sda),
-        i2c::Mode::Standard { frequency: 100000 },
-//        get_mode(),
+//        i2c::Mode::Standard { frequency: 100000 },
+        get_mode(),
         clocks,
[... cut ...]
33 |     let i2c = i2c::BlockingI2c::i2c2(                                                               
   |         ^^^ help: consider using `_i2c` instead                                                     
   |                                                                                                     
   = note: #[warn(unused_variables)] on by default                                                       
                                                                                                         
    Finished dev [unoptimized + debuginfo] target(s) in 0.13s
```

TeXitoi, 2018-10-06:
What if you add a coma after 100000? I think I've seen a bug report by the rocket author about something like that, bit I can't find it anymore.

TeXitoi, 2018-10-06:
@SergioBenitez maybe you know the issue?

dtolnay, 2018-10-06:
With a comma after 100000 it should work. This is a compiler bug: https://github.com/rust-lang/rust/issues/43081.

TeXitoi, 2018-10-06:
Oups, I didn't scroll then I checked this issue! Thanks @dtolnay.

closed 2018-10-06 by TeXitoi

johkra, 2018-10-07:
I confirm that after adding the comma the warning points to the correct line and not to the macro invocation. Glad to hear that this is a known issue and people are working on it!