rust-embedded/cortex-m-rt #130: New `entry` macro hides error locations

birkenfeld, 2018-10-18:
When using `#[entry]` on `main`, any rustc errors produced there lose location information.

It's pretty easy to work around this with an extra function `main_inner`, but this should be not too hard to solve (if I remember correctly, some recent version of `quote` allows keeping original spans?)

thejpster, 2018-10-20:
+1 for this.

japaric, 2018-10-24:
This has been reported before in https://github.com/rust-embedded/cortex-m-rt/issues/125#issuecomment-427556933 . There's a compiler bug where a missing comma can delete span information.

I have not run into this (nor seen wrong span information in GDB) in the past weeks but I always run `cargo fmt` on my crate, which adds trailing commas in several places. I don't know if `rustfmt` fixes this issue in most / all cases though (probably not). Properly fixing this in the compiler sounds like it would be difficult (https://github.com/rust-lang/rust/issues/43081#issuecomment-409694383).

> if I remember correctly, some recent version of quote allows keeping original spans?

We tried using the latest version before and didn't seem to help (see first linked issue). Also, the latest release of cortex-m-rt is already using the latest quote release.

Unfortunately, there's not much *we* can do about this issue. Perhaps add a link to the compiler bug in the documentation and suggest to run rustfmt?

birkenfeld, 2018-10-24:
Hmm, what I saw was slightly different; there was no location printed by the compiler at all, not even the location of the attribute.

But right now I can't reproduce it either, so in principle locations work and everything else is not a cortex-m-rt bug.

closed 2018-10-24 by birkenfeld