rust-embedded/cortex-m-rt #138: Links to macros broken (on docs.rs at least)

birkenfeld, 2018-10-24:
The three links to the macros in the cortex-m-rt-macros crate (here)[https://docs.rs/cortex-m-rt/0.6.5/cortex_m_rt/] are broken: they lead to e.g. https://docs.rs/cortex-m-rt/0.6.5/cortex_m_rt_macros/fn.entry.html

adamgreig, 2018-10-24:
Good spot. I'm not sure if there's a good way to cause the correct links to be generated, given they'll have to contain the version number of the cortex-m-rt-macros crate (i.e., https://docs.rs/cortex-m-rt-macros/0.1.3/cortex_m_rt_macros/attr.exception.html)? The public re-exports at the bottom of the page does have correct links for the macros.

jonas-schievink, 2019-11-26:
Hmm, I think we could put `#[doc(inline)]` on the reexport and treat `cortex-m-rt-macros` as an unstable implementation detail that users shouldn't use directly.

jonas-schievink, 2019-11-26:
Huh, looks like the current version of the macros crate doesn't even work on docs.rs: https://docs.rs/cortex-m-rt-macros/

closed 2019-11-27 by bors[bot]