rust-embedded/cortex-m-rt #145: Doesn't support JSON targets

thejpster, 2018-10-27:
If you supply a target as a JSON file, the build.rs can't detect the target. This is because `target` is a full path - in my case starts this starts `C:\`, which != `thumbv`.

I think the fix is to take the `file_stem()` of the path and use that. This should work for both JSON targets and built-in targets (that don't contain `.`). If the built-in target does use `.` (e.g. `thumbv8-m.base`) we'll need to be a bit cleverer.

japaric, 2018-10-30:
> I think the fix is to take the file_stem() of the path and use that.

This doesn't always work. There's no guarantee the file name matches the LLVM name or it's even a proper triple, e.g. `foo.json` is a valid name for a custom specification file. The proper way is to run `rustc --print cfg --target $TARGET` and then parse the output. However, on beta / stable the output doesn't contain enough information to discern ARM Linux from ARM Cortex-M. For example, `target_feature="mclass"` only appears in the nightly output of `rustc --print cfg --target thumbv7m-none-eabi` and it's not stable, i.e. it may disappear on any nightly update.

If we want to "support" custom targets the documentation should state that this is best effort support and than nightly updates may break things.

thejpster, 2018-10-31:
I agree it requires the user to name their JSON file correctly, but at least they have that option. Currently, `cortex-m` and `cortex-m-rt` don't work at all with a JSON target file (i.e. thumbv8m)

closed 2021-01-08 by bors[bot]