rust-embedded/cortex-m-rt #153: Can't compile lib on macos/nightly

little-arhat, 2018-11-12:
Hi, 

error message: 

```
LLVM ERROR: Global variable 'UserHardFault_' has an invalid section specifier '.UserHardFault': mach-o section specifier requires a segment and section separated by a comma.
error: Could not compile `cortex-m-rt`.
```

```
active toolchain
----------------

nightly-x86_64-apple-darwin (default)
rustc 1.32.0-nightly (ca79ecd69 2018-11-11)
```

`Apple LLVM version 10.0.0 (clang-1000.10.44.4)`

Full log:

```
cargo -v build
    Updating crates.io index
  Downloaded syn v0.15.20
  Downloaded quote v0.6.10
  Downloaded rand_core v0.2.2
  Downloaded proc-macro2 v0.4.23
  Downloaded rand_core v0.3.0
   Compiling proc-macro2 v0.4.23
   Compiling unicode-xid v0.1.0
   Compiling rand_core v0.3.0
   Compiling cortex-m-rt v0.6.5 (/Users/user/stuff/cortex-m-rt)
   Compiling r0 v0.2.2
     Running `rustc --crate-name build_script_build /Users/user/.cargo/registry/src/github.com-1ecc6299db9ec823/proc-macro2-0.4.23/build.rs --color always --crate-type bin --emit=dep-info,link -C debuginfo=2 --cfg 'feature="default"' --cfg 'feature="proc-macro"' -C metadata=72a72eaec06318a9 -C extra-filename=-72a72eaec06318a9 --out-dir /Users/user/stuff/cortex-m-rt/target/debug/build/proc-macro2-72a72eaec06318a9 -L dependency=/Users/user/stuff/cortex-m-rt/target/debug/deps --cap-lints allow`
     Running `rustc --crate-name unicode_xid /Users/user/.cargo/registry/src/github.com-1ecc6299db9ec823/unicode-xid-0.1.0/src/lib.rs --color always --crate-type lib --emit=dep-info,link -C debuginfo=2 --cfg 'feature="default"' -C metadata=39199b193b4b7952 -C extra-filename=-39199b193b4b7952 --out-dir /Users/user/stuff/cortex-m-rt/target/debug/deps -L dependency=/Users/user/stuff/cortex-m-rt/target/debug/deps --cap-lints allow`
     Running `rustc --crate-name rand_core /Users/user/.cargo/registry/src/github.com-1ecc6299db9ec823/rand_core-0.3.0/src/lib.rs --color always --crate-type lib --emit=dep-info,link -C debuginfo=2 -C metadata=8564cd825a7c3d9e -C extra-filename=-8564cd825a7c3d9e --out-dir /Users/user/stuff/cortex-m-rt/target/debug/deps -L dependency=/Users/user/stuff/cortex-m-rt/target/debug/deps --cap-lints allow`
     Running `rustc --crate-name build_script_build build.rs --color always --crate-type bin --emit=dep-info,link -C debuginfo=2 -C metadata=4bab656769a60c2d -C extra-filename=-4bab656769a60c2d --out-dir /Users/user/stuff/cortex-m-rt/target/debug/build/cortex-m-rt-4bab656769a60c2d -C incremental=/Users/user/stuff/cortex-m-rt/target/debug/incremental -L dependency=/Users/user/stuff/cortex-m-rt/target/debug/deps`
     Running `rustc --crate-name r0 /Users/user/.cargo/registry/src/github.com-1ecc6299db9ec823/r0-0.2.2/src/lib.rs --color always --crate-type lib --emit=dep-info,link -C debuginfo=2 -C metadata=77aa75220873f63c -C extra-filename=-77aa75220873f63c --out-dir /Users/user/stuff/cortex-m-rt/target/debug/deps -L dependency=/Users/user/stuff/cortex-m-rt/target/debug/deps --cap-lints allow`
   Compiling rand_core v0.2.2
     Running `rustc --crate-name rand_core /Users/user/.cargo/registry/src/github.com-1ecc6299db9ec823/rand_core-0.2.2/src/lib.rs --color always --crate-type lib --emit=dep-info,link -C debuginfo=2 -C metadata=cfd7cadb6afe302e -C extra-filename=-cfd7cadb6afe302e --out-dir /Users/user/stuff/cortex-m-rt/target/debug/deps -L dependency=/Users/user/stuff/cortex-m-rt/target/debug/deps --extern rand_core=/Users/user/stuff/cortex-m-rt/target/debug/deps/librand_core-8564cd825a7c3d9e.rlib --cap-lints allow`
   Compiling rand v0.5.5
     Running `rustc --crate-name rand /Users/user/.cargo/registry/src/github.com-1ecc6299db9ec823/rand-0.5.5/src/lib.rs --color always --crate-type lib --emit=dep-info,link -C debuginfo=2 -C metadata=ea0558896ffe3566 -C extra-filename=-ea0558896ffe3566 --out-dir /Users/user/stuff/cortex-m-rt/target/debug/deps -L dependency=/Users/user/stuff/cortex-m-rt/target/debug/deps --extern rand_core=/Users/user/stuff/cortex-m-rt/target/debug/deps/librand_core-cfd7cadb6afe302e.rlib --cap-lints allow`
     Running `/Users/user/stuff/cortex-m-rt/target/debug/build/cortex-m-rt-4bab656769a60c2d/build-script-build`
     Running `/Users/user/stuff/cortex-m-rt/target/debug/build/proc-macro2-72a72eaec06318a9/build-script-build`
     Running `rustc --crate-name proc_macro2 /Users/user/.cargo/registry/src/github.com-1ecc6299db9ec823/proc-macro2-0.4.23/src/lib.rs --color always --crate-type lib --emit=dep-info,link -C debuginfo=2 --cfg 'feature="default"' --cfg 'feature="proc-macro"' -C metadata=84155ba0036c210b -C extra-filename=-84155ba0036c210b --out-dir /Users/user/stuff/cortex-m-rt/target/debug/deps -L dependency=/Users/user/stuff/cortex-m-rt/target/debug/deps --extern unicode_xid=/Users/user/stuff/cortex-m-rt/target/debug/deps/libunicode_xid-39199b193b4b7952.rlib --cap-lints allow --cfg u128 --cfg use_proc_macro --cfg wrap_proc_macro`
   Compiling quote v0.6.10
     Running `rustc --crate-name quote /Users/user/.cargo/registry/src/github.com-1ecc6299db9ec823/quote-0.6.10/src/lib.rs --color always --crate-type lib --emit=dep-info,link -C debuginfo=2 --cfg 'feature="default"' --cfg 'feature="proc-macro"' --cfg 'feature="proc-macro2"' -C metadata=cd80b644755e8064 -C extra-filename=-cd80b644755e8064 --out-dir /Users/user/stuff/cortex-m-rt/target/debug/deps -L dependency=/Users/user/stuff/cortex-m-rt/target/debug/deps --extern proc_macro2=/Users/user/stuff/cortex-m-rt/target/debug/deps/libproc_macro2-84155ba0036c210b.rlib --cap-lints allow`
   Compiling syn v0.15.20
     Running `rustc --crate-name syn /Users/user/.cargo/registry/src/github.com-1ecc6299db9ec823/syn-0.15.20/src/lib.rs --color always --crate-type lib --emit=dep-info,link -C debuginfo=2 --cfg 'feature="clone-impls"' --cfg 'feature="default"' --cfg 'feature="derive"' --cfg 'feature="extra-traits"' --cfg 'feature="full"' --cfg 'feature="parsing"' --cfg 'feature="printing"' --cfg 'feature="proc-macro"' --cfg 'feature="proc-macro2"' --cfg 'feature="quote"' --cfg 'feature="visit"' -C metadata=9ce61b7cb38b1798 -C extra-filename=-9ce61b7cb38b1798 --out-dir /Users/user/stuff/cortex-m-rt/target/debug/deps -L dependency=/Users/user/stuff/cortex-m-rt/target/debug/deps --extern proc_macro2=/Users/user/stuff/cortex-m-rt/target/debug/deps/libproc_macro2-84155ba0036c210b.rlib --extern quote=/Users/user/stuff/cortex-m-rt/target/debug/deps/libquote-cd80b644755e8064.rlib --extern unicode_xid=/Users/user/stuff/cortex-m-rt/target/debug/deps/libunicode_xid-39199b193b4b7952.rlib --cap-lints allow`
   Compiling cortex-m-rt-macros v0.1.3 (/Users/user/stuff/cortex-m-rt/macros)
     Running `rustc --crate-name cortex_m_rt_macros macros/src/lib.rs --color always --crate-type proc-macro --emit=dep-info,link -C prefer-dynamic -C debuginfo=2 -C metadata=b0922efd5e82daf9 -C extra-filename=-b0922efd5e82daf9 --out-dir /Users/user/stuff/cortex-m-rt/target/debug/deps -C incremental=/Users/user/stuff/cortex-m-rt/target/debug/incremental -L dependency=/Users/user/stuff/cortex-m-rt/target/debug/deps --extern proc_macro2=/Users/user/stuff/cortex-m-rt/target/debug/deps/libproc_macro2-84155ba0036c210b.rlib --extern quote=/Users/user/stuff/cortex-m-rt/target/debug/deps/libquote-cd80b644755e8064.rlib --extern rand=/Users/user/stuff/cortex-m-rt/target/debug/deps/librand-ea0558896ffe3566.rlib --extern syn=/Users/user/stuff/cortex-m-rt/target/debug/deps/libsyn-9ce61b7cb38b1798.rlib`
     Running `rustc --crate-name cortex_m_rt src/lib.rs --color always --crate-type lib --emit=dep-info,link -C debuginfo=2 -C metadata=7a0671ac2fa964f1 -C extra-filename=-7a0671ac2fa964f1 --out-dir /Users/user/stuff/cortex-m-rt/target/debug/deps -C incremental=/Users/user/stuff/cortex-m-rt/target/debug/incremental -L dependency=/Users/user/stuff/cortex-m-rt/target/debug/deps --extern cortex_m_rt_macros=/Users/user/stuff/cortex-m-rt/target/debug/deps/libcortex_m_rt_macros-b0922efd5e82daf9.dylib --extern r0=/Users/user/stuff/cortex-m-rt/target/debug/deps/libr0-77aa75220873f63c.rlib -L /Users/user/stuff/cortex-m-rt/target/debug/build/cortex-m-rt-20aa54d9381d379f/out`
LLVM ERROR: Global variable 'UserHardFault_' has an invalid section specifier '.UserHardFault': mach-o section specifier requires a segment and section separated by a comma.
error: Could not compile `cortex-m-rt`.

Caused by:
  process didn't exit successfully: `rustc --crate-name cortex_m_rt src/lib.rs --color always --crate-type lib --emit=dep-info,link -C debuginfo=2 -C metadata=7a0671ac2fa964f1 -C extra-filename=-7a0671ac2fa964f1 --out-dir /Users/user/stuff/cortex- [... cut ...] tal=/Users/user/stuff/cortex-m-rt/target/debug/incremental -L dependency=/Users/user/stuff/cortex-m-rt/target/debug/deps --extern cortex_m_rt_macros=/Users/user/stuff/cortex-m-rt/target/debug/deps/libcortex_m_rt_macros-b0922efd5e82daf9.dylib --extern r0=/Users/user/stuff/cortex-m-rt/target/debug/deps/libr0-77aa75220873f63c.rlib -L /Users/user/stuff/cortex-m-rt/target/debug/build/cortex-m-rt-20aa54d9381d379f/out` (exit code: 101)
cargo -v build  56.20s user 13.89s system 118% cpu 58.953 total
```

adamgreig, 2018-11-12:
Hello! It looks like you might be trying to build for your native target (i.e. for OS X `apple-darwin`), but this crate only supports being built for a thumb (Cortex-M) target. Could you try `cargo build --target thumbv7em-none-eabihf` or any other target you have installed? You might need to `rustup target add thumbv7em-none-eabihf` first.

little-arhat, 2018-11-12:
@adamgreig yes, indeed, sorry! 

There was failure on ci this weekend: https://travis-ci.org/copterust/proving-ground/builds/449079670, and I've also got errors locally (see below). But indeed, after bumping some deps (panic-abort, cortex-m), today it seems to work. 

Cortex-m-rt itself also compiles fine on macos/nightly.  

```
error: aborting due to previous error

error: Could not compile cortex-m.

Caused by:
  process didn't exit successfully: rustc --crate-name build_script_build /.../.cargo/registry/src/github.com-1ecc6299db9ec823/cortex-m-0.5.8/build.rs --color always --crate-type bin --emit=dep-info,link -C opt-level=s -C codegen-units=1 -C debuginfo=2 -C debug-assertions=on --cfg 'feature="bare-metal"' --cfg 'feature="const-fn"' --cfg 'feature="inline-asm"' -C metadata=7d431aafe81dc910 -C extra-filename=-7d431aafe81dc910 --out-dir /.../target/debug/build/cortex-m-7d431aafe81dc910 -L dependency=/.../target/debug/deps --cap-lints allow (exit code: 1)
warning: build failed, waiting for other jobs to finish...
error: could not exec the linker cc
````

closed 2018-11-12 by little-arhat