rust-embedded/cortex-m-rt #157: Linker error: "no memory region specified for section '.ARM.extab'"

phil-opp, 2018-12-07:
I'm working with generators and async/await. I tried to move some more complex function into a generator, but I got the following linker error when compiling in debug mode:

```
rust-lld: error: no memory region specified for section '.ARM.extab'
```

I looked for other mentions of `.ARM.extab` on GitHub and found this PR from 2014: https://github.com/hackndev/zinc/pull/238. It seems like this section is used for unwinding and can be discarded. When I compile in release mode with LTO the error disappears, so it really seems like the section isn't needed. 

I think we could fix this problem by either discarding or including `.ARM.extab` in [link.x.in](https://github.com/rust-embedded/cortex-m-rt/blob/abed79f12921f9f6dcbaf9ad0664c5d47ebeccf8/link.x.in#L152).

My rust version is: `rustc 1.32.0-nightly (edaac35d6 2018-11-24)`.

thejpster, 2018-12-08:
If you run against a fork of cortex-m-rt and hack the link.x.in to discard the `.ARM.extab` section, does your debug build then work?

phil-opp, 2018-12-18:
@thejpster Sorry for the delay! I tried it again with a [modified link.x.in](https://github.com/phil-opp/cortex-m-rt/blob/phil-opp-patch-1/link.x.in#L153-L154), but the same error occurs. Not sure why…

phil-opp, 2019-01-21:
@thejpster Ok, I did some more investigation and was able to fix this by discarding `*(.ARM.extab.*);` instead of `*(.ARM.extab);`. I also made sure that no code is commented out this time :).

I opened a new PR in https://github.com/rust-embedded/cortex-m-rt/pull/170.

commit f7917175 (2019-01-22): Merge #170

170: Discard .ARM.extab.* r=japaric a=phil-opp

Fixes #157
PR #170 (2019-01-22): Discard .ARM.extab.*
Fixes #157
closed 2019-01-22 by bors[bot]