rust-embedded/cortex-m-rt #163: #[entry] attribute disguises application errors

therealprof, 2019-01-09:
If the function body of an `#[entry]` function contains an error the macro disguises the generated errors as if it was the cause of the problem:

```
error[E0061]: this function takes 1 parameter but 0 parameters were supplied
  --> examples/i2c_scanner.rs:30:1
   |
30 | #[entry]
   | ^ expected 1 parameter

error[E0599]: no method named `constrain` found for type `stm32f0::stm32f0x2::RCC` in the current scope
  --> examples/i2c_scanner.rs:30:1
   |
30 | #[entry]
   | ^

error[E0061]: this function takes 2 parameters but 1 parameter was supplied
  --> examples/i2c_scanner.rs:30:1
   |
30 | #[entry]
   | ^ expected 2 parameters

error[E0061]: this function takes 4 parameters but 3 parameters were supplied
  --> examples/i2c_scanner.rs:30:1
   |
30 | #[entry]
   | ^ expected 4 parameters

error: aborting due to 4 previous errors
```

japaric, 2019-01-10:
This is a compiler bug: rust-lang/rust#43081; usually caused by some trailing / missing comma: https://github.com/rust-lang/rust/issues/43081#issuecomment-409078027. It has been reported at least twice in this issue tracker, but we can't fix the problem from our side.

therealprof, 2019-01-10:
@japaric Thanks for the info. It would be great if we could link the issues together to make it easier to identify this as a known problem; I tried a query before opening this but came up empty.

japaric, 2019-01-12:
#125 and #130 are the issues I'm aware of. They ware closed by their authors so that's probably you didn't see them.

jonas-schievink, 2020-01-08:
Is there some minimal code example that reproduces this? The code linked in https://github.com/rust-embedded/cortex-m-rt/issues/125#issuecomment-427556933 doesn't compile.

Aaron1011, 2020-05-25:
This should be fixed in the latest nightly

adamgreig, 2021-07-13:
I believe this is now fixed, please re-open if not!

closed 2021-07-13 by adamgreig