rust-embedded/cortex-m-rt #188: _stext fails to move .text with lld

yodaldevoid, 2019-04-12:
I have an older program/library that I am moving to lld so that I can work with ease on more than just Linux. The code in question is https://github.com/yodaldevoid/mkl26/tree/lld_port. Please note that the example/library currently require nightly and should be built for thumbv6m-none-eabi.

To start, the code compiles and runs correctly using gcc's linker.

When attempting to compile the `blink` example with lld using `cargo build --example blink` I get the following error:
```
error: linking with `rust-lld` failed: exit code: 1
  |
  = note: "rust-lld" "-flavor" "gnu" "-L" "C:\\Users\\ga29s\\.rustup\\toolchains\\nightly-x86_64-pc-windows-msvc\\lib\\rustlib\\thumbv6m-none-eabi\\lib" "E:\\mkl26\\target\\thumbv6m-none-eabi\\debug\\examples\\blink-288c8e46abd20a15.blink.2hpe9hsl-cgu.0.rcgu.o" "-o" "E:\\mkl26\\target\\thumbv6m-none-eabi\\debug\\examples\\blink-288c8e46abd20a15" "--gc-sections" "-L" "E:\\mkl26\\target\\thumbv6m-none-eabi\\debug\\deps" "-L" "E:\\mkl26\\target\\debug\\deps" "-L" "E:\\mkl26\\target\\thumbv6m-none-eabi\\debug\\build\\mkl26-36049ce57ff99828\\out" "-L" "E:\\mkl26\\target\\thumbv6m-none-eabi\\debug\\build\\mkl26-36049ce57ff99828\\out" "-L" "E:\\mkl26\\target\\thumbv6m-none-eabi\\debug\\build\\cortex-m-rt-6e1f169148af1ece\\out" "-L" "C:\\Users\\ga29s\\.rustup\\toolchains\\nightly-x86_64-pc-windows-msvc\\lib\\rustlib\\thumbv6m-none-eabi\\lib" "-Bstatic" "C:\\Users\\ga29s\\AppData\\Local\\Temp\\rustcVhEipN\\libcortex_m_rt-193fc16ad349c2cc.rlib" "C:\\Users\\ga29s\\.rustup\\toolchains\\nightly-x86_64-pc-windows-msvc\\lib\\rustlib\\thumbv6m-none-eabi\\lib\\libcompiler_builtins-41a0f49e919d94c0.rlib" "-Tlink.x" "-Bdynamic"
  = note: rust-lld: error:
          ERROR(cortex-m-rt): The .text section can't be placed inside the .vector_table section
          Set _stext to an address greater than the end of .vector_table (See output of `nm`)

error: aborting due to previous error

error: Could not compile `mkl26`.

To learn more, run the command again with --verbose.
```

If I change line 20 in `memory.x` to `_stext = 0x410;` I then get the following error:
```
error: linking with `rust-lld` failed: exit code: 1
  |
  = note: "rust-lld" "-flavor" "gnu" "-L" "C:\\Users\\ga29s\\.rustup\\toolchains\\nightly-x86_64-pc-windows-msvc\\lib\\rustlib\\thumbv6m-none-eabi\\lib" "E:\\mkl26\\target\\thumbv6m-none-eabi\\debug\\examples\\blink-288c8e46abd20a15.blink.2hpe9hsl-cgu.0.rcgu.o" "-o" "E:\\mkl26\\target\\thumbv6m-none-eabi\\debug\\examples\\blink-288c8e46abd20a15" "--gc-sections" "-L" "E:\\mkl26\\target\\thumbv6m-none-eabi\\debug\\deps" "-L" "E:\\mkl26\\target\\debug\\deps" "-L" "E:\\mkl26\\target\\thumbv6m-none-eabi\\debug\\build\\mkl26-36049ce57ff99828\\out" "-L" "E:\\mkl26\\target\\thumbv6m-none-eabi\\debug\\build\\mkl26-36049ce57ff99828\\out" "-L" "E:\\mkl26\\target\\thumbv6m-none-eabi\\debug\\build\\cortex-m-rt-6e1f169148af1ece\\out" "-L" "C:\\Users\\ga29s\\.rustup\\toolchains\\nightly-x86_64-pc-windows-msvc\\lib\\rustlib\\thumbv6m-none-eabi\\lib" "-Bstatic" "C:\\Users\\ga29s\\AppData\\Local\\Temp\\rustcy1mc9U\\libcortex_m_rt-193fc16ad349c2cc.rlib" "C:\\Users\\ga29s\\.rustup\\toolchains\\nightly-x86_64-pc-windows-msvc\\lib\\rustlib\\thumbv6m-none-eabi\\lib\\libcompiler_builtins-41a0f49e919d94c0.rlib" "-Tlink.x" "-Bdynamic"
  = note: rust-lld: error: section .flashconfig file range overlaps with .vector_table
          >>> .flashconfig range is [0x1000, 0x100F]
          >>> .vector_table range is [0x1000, 0x10BF]

          rust-lld: error: section .flashconfig virtual address range overlaps with .vector_table
          >>> .flashconfig range is [0x0, 0xF]
          >>> .vector_table range is [0x0, 0xBF]

          rust-lld: error: section .flashconfig load address range overlaps with .vector_table
          >>> .flashconfig range is [0x0, 0xF]
          >>> .vector_table range is [0x0, 0xBF]

error: aborting due to previous error

error: Could not compile `mkl26`.

To learn more, run the command again with --verbose.
```

It appears that `.flashconfig` is not being placed at 0x400 despite what the linker script says. Am I doing something wrong here or is this some sort of bug?

yodaldevoid, 2019-04-13:
#164 seems to be related as all of this is in attempt to support the flash configuration section on an NXP device.

Disasm, 2019-04-13:
Your `.flashconfig` section definition is placed before all other sections, that's why you have some of these problems. Try setting loading address (not only virtual address) for this section.

yodaldevoid, 2019-04-13:
It is my understanding that when a load address is not given the virtual address is used as the loading address. Still, to test your suggestion I changed the `SECTIONS` command to be:
```
SECTIONS
{
    .flashconfig 0x400 : AT (0x400)
    {
        KEEP(*(.flashconfig))
        _stext = .;
    } > FLASH
}
```

With this I still see the previous suite of errors. Please correct me if I went about implementing your suggestion wrong.

Disasm, 2019-04-13:
Yeah, seems OK. I'm not sure though if it's the correct way of implementing this in case of lld. In `riscv-rt` I had to add [some dirty hacks](https://github.com/rust-embedded/riscv-rt/blob/ecc1344ffc9af1c88b3dd76b83ad56284672f888/link.x#L23-L28) to get section address working.

yodaldevoid, 2019-04-15:
Because I had a copy of the `rustc` source code laying around (and really, who doesn't?) I decided to add debug prints to `lld` and see what's going on. After poking around, it seems that `lld` is evaluating the address of the section, moving the dot there, and then proceeds to move the dot to the current position within the memory region. Because the address was evaluated before actually setting the current region, the current position in the region was not moved by the evaluation. This all means that the address evaluation is effectively ignored in favor for the current position within the region.

This all seems like a bug in `lld` to me, and so I plan on submitting a bug report and patch to fix the issue. I am currently awaiting my account getting approved as they currently have self registration on the bug tracker disabled.

TL;DR: This seems to be caused by a bug in `lld`.

yodaldevoid, 2019-04-16:
https://reviews.llvm.org/D60744

Disasm, 2019-04-16:
Thanks for fixing the issue!

yodaldevoid, 2019-04-25:
A little update on this. While my patch got merged into LLVM master, I found an issue with it shortly after. While raw binaries are created correctly, ELF binaries have the possibility of the ELF headers getting written over by a section defined in a linker script. In short, this has not been fully resolved yet.

commit d5f2a94d (2019-10-28): Rejig link.x to include more lables to help the linker lay out the objects

Tested with 1.34.0 and 1.38.0 and careful inspection of the linker map
generated on the previously failing
https://github.com/rust-lang/rust/files/3722440/minimal-rust-lld-issue.zip

Closes #188 (I believe)
Closes https://github.com/rust-lang/rust/issues/65391
commit fa056fe3 (2019-10-28): Merge #207

207: Rejig link.x to include more lables to help the linker lay out the ob… r=thejpster a=therealprof

…jects

Tested with 1.34.0 and 1.38.0 and careful inspection of the linker map
generated on the previously failing
https://github.com/rust-lang/rust/files/3722440/minimal-rust-lld-issue.zip

Closes #188 (I believe)
Closes https://github.com/rust-lang/rust/issues/65391
PR #207 (2019-10-28): Rejig link.x to include more lables to help the linker lay out the ob…
…jects

Tested with 1.34.0 and 1.38.0 and careful inspection of the linker map
generated on the previously failing
https://github.com/rust-lang/rust/files/3722440/minimal-rust-lld-issue.zip

Closes #188 (I believe)
Closes https://github.com/rust-lang/rust/issues/65391
closed 2019-10-28 by bors[bot]