rust-embedded/cortex-m-rt #189: .bss is not 4-byte aligned

lthiery, 2019-05-16:
In [this sx1276 project](https://github.com/lthiery/sx1276-rs), I'm trying to link to C code compiled using the 'cc' crate (by the way, thanks for the comment about the `fPIC` flag). 

I was able to exercise some of the functions, but as I continued to build it out, I ran into the following error:
```
          BUG(cortex-m-rt): .bss is not 4-byte aligned
          
          rust-lld: error: 
          BUG(cortex-m-rt): start of .heap is not 4-byte aligned
```

I assume my static C-library has brought something in that misaligns things, but I'm confused how the following doesn't more or less guarantee alignment:

```
  /* ### .bss */
  .bss : ALIGN(4)
  {
    *(.bss .bss.*);

    . = ALIGN(4); /* 4-byte align the end (VMA) of this section */
  } > RAM
```
I've tried adding `__attribute__ ((aligned (8)))` to no avail. 

Wonder if anyone has ideas? Hack fixes welcome as much as nice fixes :)

thejpster, 2019-06-04:
I tried to recreate with some weird sized globals but couldn't. Do you have a minimal reproducer?

lthiery, 2019-06-04:
I wasn't able to make a minimal reproducer and I tried to hack a fix by adding weird sized globals but that didn't fix it either. 

I did test pull request #190 and it did fix my issue though!

NickeZ, 2019-06-05:
The fix hasn't been merged yet, so I think you should reopen this issue.

commit 3d4e730a (2019-06-08): Merge #190

190: Align sections r=korken89 a=NickeZ

Hey, I had the same issue as in #189. Don't ask me why this works I looked at other linker scripts.

Fixes #189
PR #190 (2019-06-08): Align sections
Hey, I had the same issue as in #189. Don't ask me why this works I looked at other linker scripts.

Fixes #189
closed 2019-06-08 by bors[bot]