rust-embedded/cortex-m-rt #191: Possible bug in startup code for Cortex-M0

jfrimmel, 2019-06-03:
Hello,

maybe I have found a bug using this crate. Normally I would say, that the problem is on my side, but I reduced the case as much as I could and the error still arose. I cannot find a user-error in this reduced code, but I may be wrong. So, whats the bug?

When using a large zero-initialized array, the generated startup code is incorrect. Dependend on the optimization setting, the `zero_bss` code will either loop infinite, produce a HardFault or will work! 

I have created a repository with a minimal working example and steps to reproduce:
[jfrimmel/cortex-m-rt-bug-MWE](https://github.com/jfrimmel/cortex-m-rt-bug-MWE).

I'm happy to provide more information, if necessary.

adamgreig, 2019-06-16:
Hi, thanks for reporting and providing a MWE!

I had a go with your MWE on an STM32F0 with just one change (`memory.x` flash address updated to `0x08000000` for this MCU) it seems to behave as expected in debug mode -- that is, it correctly zeros bss, main runs to increment each element of `GLOBAL_ARRAY`, and then sits in the final `loop{}`.

```
(gdb) r
The program being debugged has been started already.
Start it from the beginning? (y or n) y
Starting program: /home/adam/tmp/cortex-m-rt-bug-MWE/target/thumbv6m-none-eabi/debug/cortex-m-rt-bug-mwe
^C
Program received signal SIGINT, Interrupt.
main () at src/main.rs:20
20	    loop {}
(gdb) p GLOBAL_ARRAY
$6 = [1 <repeats 1024 times>]
```

Compared to what you note in your `README`, I see a sensible address for sbss:

```
(gdb) b zero_bss
Breakpoint 2 at 0x8000b7c: file /home/adam/.cargo/registry/src/github.com-1ecc6299db9ec823/r0-0.2.2/src/lib.rs, line 171.
(gdb) r
The program being debugged has been started already.
Start it from the beginning? (y or n) y
Starting program: /home/adam/tmp/cortex-m-rt-bug-MWE/target/thumbv6m-none-eabi/debug/cortex-m-rt-bug-mwe

Breakpoint 2, r0::zero_bss (sbss=0x20000000 <cortex_m_rt_bug_mwe::GLOBAL_ARRAY>, ebss=0x20000400)
    at /home/adam/.cargo/registry/src/github.com-1ecc6299db9ec823/r0-0.2.2/src/lib.rs:171
171	    while sbss < ebss {
(gdb) p sbss
$3 = (u32 *) 0x20000000 <cortex_m_rt_bug_mwe::GLOBAL_ARRAY>
(gdb) p ebss
$4 = (u32 *) 0x20000400
(gdb) p &sbss
$5 = (u32 **) 0x200007f0
```

Whereas you apparently see `&sbss` as _inside_ `GLOBAL_ARRAY`. Are you absolutely sure you printed `&sbss` instead of `sbss` at that point? Since `sbss` is updated to point to each address in bss in turn (check out `zero_bss` [here](https://github.com/japaric/r0/blob/master/src/lib.rs#L174)), it absolutely makes sense to see `sbss` as `0x200003f0` but would be a bizarre error for the stack-allocated function variable to be allocated inside what the linker already knows is an allocated object.

You might try sticking `cortex_m::asm::nop()` inside your `loop{}` since there is a known issue with LLVM mishandling empty loops. You could also try using `arm-none-eabi-gcc` as the linker (put `"-C", "linker=arm-none-eabi-gcc"` in your `.cargo/config`) and see if that makes any difference.

jfrimmel, 2019-06-23:
First, excuse me for the long delay, but I was very busy with other projects.

I had some more time to investigate this issue on my embedded system. I'm very certain, that the observed behavior is __not__ due to this crate. I never really thought of this, as the Rust code looked correct at all time. Below you'll find my debugging steps, maybe they will be helpful in future issues of this kind.

So, first of all, where was the error? To answer the question I should add the used microcontroller: it was a EFM32TG11B120, which is a relatively new controller by SiLabs. This controller comes equipped with an internal bootloader, that runs first after the reset, i.e. before the first instruction of the flashed application. This bootloader __sets up the stack pointer incorrectly!__

The stack pointer was set to a value `1kiB` above the RAM start. So the stack pointer could eventually reach the bottom of the RAM when initializing the array with `1kiB` size. That has hidden the error, when only using little memory, since the stack pointer was always inside the RAM (although it could override user data, which almost certainly happend in my application before...).
Note, that the stack pointer value, i.e. the first entry in the "vector table" was specified correctly to the top of the RAM, but the bootloader didn't set this value up before calling the application.

The solution (for me) was to disable the bootloader via the `MSC` unit. Refer to the datasheet how to do this or reach out to me if you have problems doing that. After disabling the bootloader the stack pointer was set up correctly by the Cortex-M core and everything worked as expected.

I'm not sure if this is a real error of the bootloader or again an error in its usage. But I think that sharing the solution might help someone out there :smile:

adamgreig, 2019-06-30:
Thanks for wrapping this issue up!

closed 2019-06-30 by adamgreig