rust-embedded/cortex-m-rt #208: Unable to compile under x86_64-unknown-linux-musl

mamins1376, 2019-10-25:
Coming from The Embedded Book, I'm stuck in [the cross-compiling section](https://rust-embedded.github.io/book/start/qemu.html#cross-compiling).

```
# cargo build --target=thumbv6m-none-eabi
error: cannot produce proc-macro for `cortex-m-rt-macros v0.1.5` as the target `thumbv6m-none-eabi` does not support these crate types
```

Also tried to do that with both `RUSTFLAGS='-C target-feature=-crt-static'` and adding that to `.cargo/config`. I think there supposed to be a way to the that specifically to `rustc` when compiling `cortex-m-rt-macros`?

thejpster, 2019-10-28:
I think this is related to https://github.com/rust-lang/rust/issues/40174.

mati865, 2019-11-13:
It's not related.
`thumbv6m-none-eabi` doesn't support dynamic linking at all and proc-macros are just shared libraries. I'm not aware of any way to build proc-macros for that target.

jonas-schievink, 2020-08-30:
The upstream issue has been fixed, closing.

closed 2020-08-30 by jonas-schievink