rust-embedded/cortex-m-rt #214: #[interrupt] should check that the interrupt exists on the target

jonas-schievink, 2019-11-21:
In reference to this comment:

https://github.com/rust-embedded/cortex-m-rt/blob/7050c73c428e8ebab7a78d8e22f3214ebf14e7dd/macros/src/lib.rs#L289-L290

I've lost a lot of time due to not realizing that `MemoryManagement` does not exist on M0(+) chips (instead, `HardFault` is invoked). At least in this particular case it should be easy to forbid it since it never exists on `thumbv6`-based targets.

TODO: Find all other interrupts that don't always exist.

jonas-schievink, 2021-01-26:
Looks like this is currently unimplementable: Proc macros have no access to the build target, for some reason (`TARGET` is unset), and there are no `#[cfg]` flags we can use to detect a specific ARM version.

adamgreig, 2021-01-26:
I guess the proc macro is being built for the host so it makes some sense there's no target. Could we use cfg flags set from build.rs or are they only visible to the parent crate?

jonas-schievink, 2021-01-26:
They're only visible to the crate that owns the build script, I think

adamgreig, 2021-01-26:
Could we do a similar trick to interrupts and generate a dummy reference to the named exception from c-m-rt which will error if the variant doesn't exist there?

jonas-schievink, 2021-01-26:
Yes, I think that should work!

closed 2021-01-26 by bors[bot]