rust-embedded/cortex-m-rt #225: #[interrupt] and #[entry] attributes fail clippy::missing_safety_doc lint

Nemo157, 2019-12-22:
```
error: unsafe function's docs miss `# Safety` section
  --> embrio-nrf51/src/lib.rs:99:1
   |
99 | #[interrupt]
   | ^^^^^^^^^^^^
   |
   = help: for further information visit https://rust-lang.github.io/rust-clippy/master/index.html#missing_safety_doc
```

```
error: unsafe function's docs miss `# Safety` section
  --> examples/pca10031/examples/hello.rs:10:1
   |
10 | #[entry]
   | ^^^^^^^^
   |
   = note: `-D clippy::missing-safety-doc` implied by `-D warnings`
   = help: for further information visit https://rust-lang.github.io/rust-clippy/master/index.html#missing_safety_doc
```

Using `cortex-m@0.6.1`, `cortex-m-rt@0.6.11` and `cortex-m-rt-macros@0.1.7` via `nrf51@0.7.0`.

Nemo157, 2019-12-22:
At least in `clippy 0.0.212 (69f99e74 2019-12-14)` `missing_safety_doc` is a default `warn` lint, so would be nice to have it passing (either by adding a note or `allow`ing it in the generated code).

jonas-schievink, 2019-12-22:
Let's just allow the lint then. These are internal functions that should not be called by the user.

jonas-schievink, 2019-12-22:
Oh, this looks like a bug in Clippy, actually. It shouldn't warn about stuff a user can't control.

Nemo157, 2019-12-24:
For a little bit extra context (mostly for the clippy issue I'll link here in a sec) the expanded code looks like:

```rust
#[export_name = "UART0"]
pub unsafe extern "C" fn __cortex_m_rt_UART0_trampoline() {
    __cortex_m_rt_UART0()
}
fn __cortex_m_rt_UART0() {
    {
        extern crate cortex_m_rt;
        interrupt::UART0;
    }
    // ...interrupt body...
}
```

```rust
#[export_name = "main"]
pub unsafe extern "C" fn __cortex_m_rt_main_trampoline() {
    __cortex_m_rt_main()
}
fn __cortex_m_rt_main() -> ! {
    // ...entry body...
}
```

jonas-schievink, 2020-01-04:
rust-lang/rust-clippy#4949 has been fixed, closing

closed 2020-01-04 by jonas-schievink