rust-embedded/cortex-m-rt #231: Rename `#[interrupt]` to `#[interrupt_helper]`

jonas-schievink, 2020-01-06:
`#[interrupt]` is not meant to be used directly. Instead, it is reexported by PACs, which also have to export the `interrupt` enum to make the macro work.

Since `#[interrupt]` is located right next to `#[exception]`, `#[entry]`, and `#[pre_init]`, which *do* work from inside `cortex_m_rt`, it's very easy to accidentally import it from the wrong place. Renaming it to `#[interrupt_helper]` and reexporting it via `pub use cortex_m_rt::interrupt_helper as interrupt;` prevents that.

This of course comes with one big downside: It breaks every single PAC. Since they are generated via svd2rust, this isn't a terrible fix, but it will still take time to propagate. An svd2rust release should be coordinated to be available at the same time this rename is shipped (if we decide to do this).

adamgreig, 2020-01-06:
If we want to make it more clearly not-for-general-use I'd even call it `#[__interrupt]` or `#[__interrupt_helper]` or something. But I'm not sure this really solves a problem - are people in the wild running into errors doing this, enough to be worth breaking everything?  It would still be listed as `interrupt_helper` next to `entry` and `pre_init`, so seems about as likely to get used incorrectly while someone sighs about the annoying name or whatever.

> Since they are generated via svd2rust

I wonder if everyone using cortex-m-rt is also using an svd2rust PAC; I know I often don't but I'm not sure about the various other embedded projects (tock, drone, bobbin, etc).

jonas-schievink, 2020-01-06:
> are people in the wild running into errors doing this, enough to be worth breaking everything?

I ran into this fairly recently. It's definitely not worth to issue a breaking change just because of this rename alone, but I have other breaking changes of similar magnitude in mind that can be done at the same time.

> I wonder if everyone using cortex-m-rt is also using an svd2rust PAC

Good point. I don't think we have any concrete numbers on this, but I'd imagine that most people use an svd2rust PAC, and the rest use either a hand-written or otherwise generated register access crate.

thejpster, 2020-01-14:
I'm fine with this, but iff we have other breaking changes going in.

jonas-schievink, 2021-02-02:
Closing in favor of https://github.com/rust-embedded/cortex-m-rt/issues/238#issuecomment-595332945

closed 2021-02-02 by jonas-schievink