rust-embedded/cortex-m-rt #237: Make the `HardFault` trampoline opt-in

jonas-schievink, 2020-01-09:
Currently, *any* user of this crate has to pay for a small Assembly trampoline that provides access to the `&ExceptionFrame` to the `HardFault` handler (even if they don't use the `&ExceptionFrame`, or don't even define any `HardFault` handler):

https://github.com/rust-embedded/cortex-m-rt/blob/492c7bf6212486b378ecf0518572d180a50c83a1/asm.s#L8-L19

Since few people make use of the `&ExceptionFrame` argument, we could add a Cargo feature to turn it on instead, and generate the right linker script in `build.rs`. By forwarding the feature to the macro crate, `#[exception]` can provide good diagnostics when trying to use `&ExceptionFrame` without the feature enabled.

I am planning on extending the trampoline mechanism to other faults and maybe even all exceptions, which I've prototyped by making the ASM trampoline generic and have it work for all faults. The trampoline code could also benefit from ARMv7/v8-specific improvements, which I'll also work on, which should shrink it again (for the higher platforms at least).

I suppose we could also have one Cargo feature per exception. The build script would then generate a linker script that pulls in exactly the trampolines that are needed. This should even still work with the precompiled `.a` files, if we do it right (putting every trampoline in its own section and making sure they're only pulled in when necessary).

Initially, defining 10 or 11 Cargo features for this sounds pretty sketchy, but most apps won't define any of them, and even "advanced" apps that implement an RTOS can get away with just 1 or 2 (one for SVCall to read syscall arguments, and one on HardFault/DefaultHandler for debugging crashes).

An ideal solution would not require Cargo features at all and instead make the `#[exception]` macro pull in the Assembly trampoline. I have not found any way to do this though. If one does come up (perhaps using the currently unstable [`#[link_args]`](https://github.com/rust-lang/rust/issues/29596) attribute), we can make the Cargo features no-ops until the next major version.

adamgreig, 2020-02-04:
From the lack of comments I guess this isn't too objectionable to anyone!

I don't think the 24 byte flash overhead is a huge price to pay as-is, and definitely not fussed about runtime cost of the trampoline, but I agree it's very rarely used so if we can totally eliminate it for most users and leave it feature gated for users who need it, that seems fine.

Having it available on SVCall and a few other faults seems useful too; what is the cost implication? Presumably a generic trampoline will only be in flash once, but I guess there's a small runtime overhead to all the faults and exceptions it intercepts, which might be fine for HardFault but perhaps less so for other more commonly used exceptions.

jonas-schievink, 2020-02-05:
Yeah, that's right. The generic trampoline is also larger than the specialized one, and much more complicated to actually write/maintain.

jonas-schievink, 2020-02-05:
We could also start by adding Cargo features only for `NMI`, `HardFault`, `SVCall` and `PendSV`, which are supported on all architectures we target and should cover most use cases. Then we'd only expose 4 Cargo features instead of... however many exceptions there are on thumbv7/thumbv8.

Dirbaio, 2021-04-20:
I just had to fight CMRT to workaround the mandatory trampoline because I needed access to more things (like r4-r11), so +1 to make it optional.

Alternative idea: why not completely remove the trampoline in CMRT (so HardFault works like other eceptions), and then create a new crate with exception frame manipulation functionality?

The current ExceptionFrame functionality feels like "porcelain" to me (see #250), and quite opinionated (read-only access, no access to r4-r11, no way to return to a custom EXC_RETURN value (the magic 0xFFFFFFFD addresses in LR). It would be better suited for another crate, with probably expanded functionality and customizability.