rust-embedded/cortex-m-rt #255: Add an INFO section in the linkerscript for instrumentation

korken89, 2020-03-27:
To be able to place strings and similar in areas of the ELF file that will not be loaded, eg for the use in instrumentation tools, an INFO marked section is need.
I propose we define a section for this, along the lines of the following:

```
SECTIONS {
  ...
  .info_section (INFO) :
  {
    *(.info_section .info_section.*);
  }
  ...
}
```

Exact section name can be bikeshed.

jonas-schievink, 2020-03-27:
Isn't `.note` meant for something like this?

korken89, 2020-03-27:
That is a good name as well.

japaric, 2020-03-31:
1. I'd be wary of adding a non-standard section with no defined semantics. It seems like it would be very easy for it to end up with unrelated things that break the semantics of other things (addresses of symbols, order between symbols, etc.). A counter-example of this point is `.uninit`: it is not standard but has well defined semantics: it holds uninitialized data (data that won't be zeroed or initialized before main).

2. If you want custom sections for whatever reason you can always just tack an extra linker script to your linker invocation. [`stlog`] does this.

[`stlog`]: https://docs.rs/stlog/0.3.3/stlog/#requirements

3. `.note` is a standard section with defined sections (see the ELF spec; there's a type reserved for it: `SHT_NOTE`).

korken89, 2020-04-01:
I agree on your worry in principle, however I strongly believe that we need to cater to both 1) developers on the firmwares and 2) the developers of the tools from the base ABI so people wanting to create tools for us has a stable base to lean on. As it is today we are leaving (2) half-way there.
If one needs to always specify a custom linker script there will never be compatibility between tools, which is unacceptable if you ask me. 

However I am open to other ideas, what would be your recommendation to provide data from the firmware side to the host side in the binary without impacting the size of the final binary other than having an `INFO` section?

On `.note`, I missed that is was an existing section.