rust-embedded/cortex-m-rt #267: linker not placing __ebss at the end of .bss

Dirbaio, 2020-04-29:
This is happening to me when linking with C code. C code is compiled with clang 10 into a static lib and then linked with the crate. This causes the C code to break because globals are filled with garbage on boot, instead of zerofilled.

```
[dirbaio@mars firmware]$ readelf -s target/thumbv7em-none-eabihf/debug/firmware2
   [snip]
  1851: 20020334     0 NOTYPE  GLOBAL DEFAULT   16 __ebss
   [snip]

[dirbaio@mars firmware]$ readelf target/thumbv7em-none-eabihf/debug/firmware2 -S

Section Headers:
  [Nr] Name              Type            Addr     Off    Size   ES Flg Lk Inf Al
   [snip]
  [16] .bss              NOBITS          20011e08 01ae08 00e5b2 00  WA  0   0  8
   [snip]
```

As you can see, .bss is 0x20011e08-0x2002cc10 but __ebss is at 0x20020334, in the middle of it.

rustc version: 1.44.0-nightly (b2e36e6c2 2020-04-22)

Dirbaio, 2020-04-30:
It seems it has to do with `INSERT AFTER` (see #266). I can't reproduce it without it.  I also found that defining __ebss outside the section fixes it.

rafaelcaricio, 2020-06-14:
Thanks @Dirbaio ! I was having the same issue as you and it's now working for me after I made the change you mentioned ("defining __ebss outside the section").

commit 092f6a28 (2020-08-29): Merge #287

287: Fix common uses of INSERT AFTER with .bss and .text r=adamgreig a=mattico

Fixes #267 
Fixes #266

This fixes two related issues.

1. Named sections are often inserted after `.bss` or `.text` in order to have them handled as if they were part of that section. Defining the start/end symbols outside of the section allows this to work.
2. Uninitialized C statics will end up as common symbols which end up in the COMMON input section. If this section is orphaned, it will likely end up placed after `.bss`. C code often expects these statics to be zero initialized. The first change w [... cut]
PR #287 (2020-08-29): Fix common uses of INSERT AFTER with .bss and .text
Fixes #267 
Fixes #266

This fixes two related issues.

1. Named sections are often inserted after `.bss` or `.text` in order to have them handled as if they were part of that section. Defining the start/end symbols outside of the section allows this to work.
2. Uninitialized C statics will end up as common symbols which end up in the COMMON input section. If this section is orphaned, it will likely end up placed after `.bss`. C code often expects these statics to be zero initialized. The first change would cause these symbols to be placed be [... cut]
closed 2020-08-29 by bors[bot]