rust-embedded/cortex-m-rt #269: HardFault == panic sets the correctness bar too high for panic handlers

japaric, 2020-05-19:
as pointed out in rust-embedded/cortex-m#196 interrupt::free does not mask NMI or HardFault meaning that panic handlers that do this:

``` rust
fn panic(_: &PanicInfo) -> ! {
   interrupt::free(|_| unsafe {
       // access static mut variables
    });
}
```

become unsound after #258. Those panic handlers assume no reentrancy but calling panic from HardFault or NMI can result in the panic handler being reentered which would be UB. This is likely going to make panic handlers that contain RAM buffers, like panic-rtt and panic-ram, unsound.

The same issue extends to DefaultHandler, I think, because the NMI will be serviced by DefaultHandler and, e.g. a Debug implementation could call `nmipendset` which is a safe operation (or at least, like pending any other interrupt, should be safe operation; I don't think we expose a safe API for it ATM though)

Basically, with this change only panic handlers that don't touch memory (e.g. panic-loop) or implement fully lock-free algorithms using atomics (I'm not aware of any) will remain sound.

cc @korken89

adamgreig, 2020-05-19:
At the moment the user can already safely call `panic!()` inside their own HardFault anyway (hence the existence of https://github.com/rust-embedded/cortex-m/issues/196) so the change here doesn't alter the status quo as I understand it; it's just something we'd want to remedy at the same time as https://github.com/rust-embedded/cortex-m/issues/196 if possible, right?

japaric, 2020-05-19:
> it's just something we'd want to remedy at the same time as rust-embedded/cortex-m#196 if possible, right?

not necessarily. If the solution to 196 is to make custom HardFault unsafe then PR #258  still makes panic-rtt and similar UB without the user setting any custom HardFault handler

the option of making interrupt::free unsafe I don't see as a viable solution for panic handlers because they can be called from any context including HardFault and NMI -- the implementer of the panic handler may have no way to know whether it's being called from such a context thus it can't enforce the safety requirement of interrupt::free

japaric, 2020-05-19:
> not necessarily

sorry, incomplete sentence / idea. I mean to say that "fixing" 196 does not necessarily mean that PR 258 != UB for existing panic handlers. Hence, I think we should not make the default UB, even if end users can UB themselves, until 196 is addressed.

adamgreig, 2020-05-19:
(I think you mean #257 rather than 258).

Sure, I just mean we probably want to work out the solution to the overall problem at the same time and fix both issues at once; I appreciate we might need to back off 257 as part of solving 196 (though I really hope we can avoid doing so!).

closed 2020-08-23 by bors[bot]