rust-embedded/cortex-m-rt #274: How to link exceptions to C handlers

coherent-lyn, 2020-06-23:
I have a number of exception handlers in a C library.

How can I populate the vector table with these functions?

adamgreig, 2020-07-09:
The easy way to do this is to call your C handler from a suitably annotated Rust function, e.g.

```rust
extern "C" { fn my_spi1_handler(); }

#[interrupt(SPI1)]
fn spi1() {
    unsafe { my_spi1_handler(); }
}
```

If the function call overhead is not acceptable and you can change the name of your C methods, you should find you can also just name them with the interrupt name (i.e. whatever's in the Interrupt enum in your PAC, `SPI1` in the example above) and they'll be placed into the vector table for you.

If you can't change their name and can't afford the function call, you may be able to extend the linker script to use your C functions as handlers with `PROVIDE(SPI1 = my_spi1_handler)`, but I haven't tested that idea. Otherwise you'll have to modify your PAC where it defines `__INTERRUPTS` to point to your C function instead of its default names.

jonas-schievink, 2021-01-08:
Closing as this question was answered.

closed 2021-01-08 by jonas-schievink