rust-embedded/cortex-m-rt #281: How to utilize CCM SRAM for uninit statics?

tstellanova, 2020-07-24:
The documentation in `lib.rs` mentions using CCRAM for `_stack_start` but it does not seem possible to use CCM for eg uninit statics because there is no section in [link.x.in](https://github.com/rust-embedded/cortex-m-rt/blob/master/link.x.in#L60) that directs its output to CCM.  

I realize not all cortex-m devices have something like CCM, but having a placeholder section available in `link.x` for that, when it is available, would allow developers to take advantage of it when it is available.

adamgreig, 2020-07-24:
You can do this without any changes to cortex-m-rt, though perhaps it could be better documented.

In your `memory.x`, write something like:

```
MEMORY
{
    FLASH  (rx) : ORIGIN = 0x08000000, LENGTH = 1024K
    RAM    (rw) : ORIGIN = 0x20000000, LENGTH = 128K
    CCMRAM (rw) : ORIGIN = 0x10000000, LENGTH = 64K
}

SECTIONS
{
    .ccmram (NOLOAD) : ALIGN(4)
    {
        *(.ccmram .ccmram.*);
        . = ALIGN(4);
    } > CCMRAM
} INSERT AFTER .bss;

/* Place stack in CCMRAM (optional) */
_stack_start = ORIGIN(CCMRAM) + LENGTH(CCMRAM);
```

You can then place uninit statics into CCMRAM:

```rust
#[link_section=".ccmram"]
static mut BUF: [u8; 1024] = [0u8; 1024];
```

tstellanova, 2020-07-24:
Thanks for the assistance-- I did not realize SECTIONS could be modified this way.  This works!

closed 2020-07-24 by tstellanova

therealprof, 2020-07-28:
Is this documented somewhere in this detail? If not it probably should be.

tstellanova, 2020-07-28:
> Is this documented somewhere in this detail? If not it probably should be.

I'd be happy to write this up somewhere.  Where's the right spot?  Maybe the lib [comments](https://github.com/rust-embedded/cortex-m-rt/blob/98e4c3c8597f1dfa52d55616c7ce7cf90682d95b/src/lib.rs#L52)