rust-embedded/cortex-m-rt #298: undefined reference to `__sheap'

huntc, 2020-10-14:
When linking my library with the following:

```rust
    unsafe { ALLOCATOR.init(cortex_m_rt::heap_start() as usize, HEAP_SIZE_BYTES) }
```

I receive a:

```
undefined reference to `__sheap'
```

I didn't see a means by which this symbol is declared. Any pointers? I'm linking my Rust library to an nRF52 C based project. Thanks.

thalesfragoso, 2020-10-14:
Do you have this line on your `.cargo/config` ?

https://github.com/rust-embedded/cortex-m-quickstart/blob/3cbcb9956acd638c0fb864937770f51d663fba6f/.cargo/config#L18

huntc, 2020-10-14:
I don’t have that line. Also, I’m using the gcc linker so I might have to read up on those options. Is there a section in the doc discussing the need for these options? Happy to PR the doc once I learn more if not. Thanks.

huntc, 2020-10-14:
Unfortunately, that line, or a combo of the other lines doesn’t help to resolve the symbol.

adamgreig, 2020-10-14:
`__sheap` is defined in the linker script [here](https://github.com/rust-embedded/cortex-m-rt/blob/master/link.x.in#L162); if you're not getting it it probably means the linker script is not being included. Regardless of gcc or lld linker, you need to pass `-Tlink.x` somehow or other, which is what that line in `.cargo/config` does. If you have a RUSTFLAGS environment variable set, it will override `.cargo/config`, or perhaps you're specifying another linker script somehow?

huntc, 2020-10-14:
Thanks. I’m getting the hang of this now. My situation is that I’ve got another nRF52 SDK C based project actually linking this Rust-based library statically. The nRF project is responsible for doing the linking and has a linker script already. So what’s the approach here… do I need to include just that one symbol from link.x, or am I looking to somehow merge link.x with my existing linker script?

huntc, 2020-10-15:
It occurs to me that I might not even need the cortex-m-rt library given that I’m calling Rust as a static lib and my nRF52 project has effectively set up its runtime. Obviously, I’ll need to take care of setting up an allocator etc… or am I in for a world of pain assuming an approach without cortex-m-rt?

hannobraun, 2020-10-15:
I have little experience with such C/Rust integrations, but I'd assume you're in a world of pain if you try to use cortex-m-rt _and_ have some C-based runtime. cortex-m-rt will set up a vector table, do all the setup needed before entering `main` (copying/zeroing memory), etc. Which is all stuff that your C runtime presumable already does.

It's required for Rust applications. For a Rust library you only need the cortex-m crate, which will provide you some APIs, but won't interfere with another program that calls into your Rust code.

Sympatron, 2020-10-15:
But I suppose @huntc would still have to define a global allocator if the Rust library depends on it. Is there some allocator one could use that uses `malloc` and `free` provided by the C runtime? Otherwise you would have to define 2 separate heaps, which is probably not desirable.

huntc, 2020-10-16:
Thanks for the replies, everyone. I'll keep this issue open as I'd like to ultimately raise a PR to improve the doc around calling static libs in an embedded context.

Meanwhile, I did provide my own allocator and things appear to work ([although I'm having difficulty getting tests to work](https://users.rust-lang.org/t/running-doc-and-unit-tests-for-nostd/50179)). Here's the gist of it - happy for feedback on the gist as I'm a Rust newbie:

https://gist.github.com/huntc/ab9a505683647aac7bccd2df0fc75f9e)

Thanks!

adamgreig, 2021-07-13:
I'm closing this issue while doing some backlog triage; please feel free to re-open if you do get back to documenting calling static libs in an embedded context, that would definitely be useful! Perhaps it could go into the [Rust with C](https://github.com/rust-embedded/book/blob/master/src/interoperability/rust-with-c.md) chapter of the book?

closed 2021-07-13 by adamgreig