rust-embedded/cortex-m-rt #322: Position Independent Code

noonien, 2021-04-02:
Hello!

From what I can gather, cortex-m-rt doesn't currently support being compiled with position independent code.

I would like to have multiple applications on the flash, and have the bootloader chooses which one to run by just changing the VTOR.

Is it possible to have this work?

adamgreig, 2021-04-02:
Unfortunately Rust doesn't really support this for the thumb targets yet; there's some more info here:

* https://github.com/rust-lang/rust/issues/54431
* https://freenode.logbot.info/rust-embedded/20210214#c6910695
* #320 

It might work for some limited applications if vtables and perhaps statics weren't used, but in general it's not currently supported. You could possibly imagine the bootloader doing some runtime patching or moving code around to make it work, but it wouldn't be simple.

noonien, 2021-04-02:
I see, thanks for the links!
I'll follow rust-lang/rust#54431.

closed 2021-04-02 by noonien